commaai/openpilot #21707: 0.8.6 update on RAV4 2020 hybrid

amsalp, 2021-07-24:
Aggressive steering and oversteer over speeds and slow recovery.

The steering feels like it lets go and recovers halfway around a corner. Like a hold then release type scenario. 

**Device/Version information:**
 - Device: [comma two]
 - Dongle ID: [bde2c8ra474e7a8a]
 - Route: [2021-07-23--08-35-40]
 - Timestamp: [from start to finish]
 - Version: [release 2 (0.8.6)]
 - Car make/model: [Toyota Rav4 2020 hybrid ]
0.8.5 was working really well

HaraldSchafer, 2021-07-24:
I'll look into this tomorrow!

amsalp, 2021-07-25:
Ok no problem I've added more drives after this and tried laneless. I seems a little better. But I does cut corners and recovers very aggressively. Some parts it makes me a little seasick when it does it's overcorrection. Wobbles the car. I all so  hear the break relay clicking when it comes to a stop behind another car.

yufeng66, 2021-07-25:
My eps firm version is 8965B42171. I think neither of the two PID tuning works very well for me with the laneless model at this point.  The more relaxed one does slow and subtle ping pong on highway.  It is tolerable and car never go over the lane line. The more aggressive one got more noticeable ping pong at local speed on big curves. It is less tolerable. The steering wheel would turn too fast and make the experience less chill and almost unsafe. I think a new tuning might be needed. I try to use INDI a bit and liked result so far. The following is what I did. basically the Prius tuning with higher gain. 

      ret.lateralTuning.init('indi')
      ret.lateralTuning.indi.innerLoopGainBP = [0.]
      ret.lateralTuning.indi.innerLoopGainV = [6.0]
      ret.lateralTuning.indi.outerLoopGainBP = [0.]
      ret.lateralTuning.indi.outerLoopGainV = [4.0]
      ret.lateralTuning.indi.timeConstantBP = [0.]
      ret.lateralTuning.indi.timeConstantV = [1.0]
      ret.lateralTuning.indi.actuatorEffectivenessBP = [0.]
      ret.lateralTuning.indi.actuatorEffectivenessV = [1.0]
      ret.steerActuatorDelay = 0.2

HaraldSchafer, 2021-07-25:
This doesn't really have anything to do with laneless. Laneless may be smoother in general, but the lateral tuning of the car is very poor. You can see in the plot the car's steering angle seems to go into oscillation at 34372 seconds when that was never commanded by the planner.

[image]

My best guess is this commit broke things for your car. 
https://github.com/commaai/openpilot/commit/f40e75edf83e3e0e224583cd9b6cf865c1cb4a2d#diff-c933b99dbed0471164bc0132e6c5dcb8d32ce3e32051cf1309d60c6474c1c332

adeebshihadeh, 2021-07-25:
Fixed in #21715

closed 2021-07-25 by adeebshihadeh