commaai/openpilot #21708: Pushing Brake causes multiple and quick errors on dash

williefeb19, 2021-07-24:
**Describe the bug**
While OP is engaged and following a lead vehicle if the brake pedal is pressed while the Adaptive Cruse Control (ACC) is braking there will be a rapid display of errors and warning tones then OP will disengage as expected. OP is able to be re-engaged without any issue. 

**How to reproduce or log data**
With ACC enabled and following a lead car press the brake pedal when ACC either starts to brake or is already braking. There will be multiple error that come up on the dashboard rapidly then clear.

There is no issue if the brake pedal is pressed with no lead car ACC disengages and OP disables as expected.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: d60b0e20649547c2
 - Route: d60b0e20649547c2|2021-07-24--15-54-27
 - Timestamp: 11min (16:05cst)
 - Version: release2 (0.8.6)
 - Car make/model: Toyota Camry XSE 2018

**Additional context**

nanshao133, 2021-08-05:
我的2018款凯美瑞混动版也是同样的情况，从081到086版本都存在急刹车时原车仪表盘警告信，和提醒：雷达巡航不可用的提醒

adeebshihadeh, 2021-09-02:
How long do you see the warnings and faults? Is it around a second or is it much longer?

williefeb19, 2021-09-03:
The warnings/faults only last about a second. I have attached a video clip (this is from a different time then the trip listed above but same result)

https://user-images.githubusercontent.com/55070773/131945510-578b2db3-d2bd-47ec-b05e-c64fbd789166.mp4

adeebshihadeh, 2021-09-04:
Our Rav4 does something similar (the UI is a bit different), and this is somewhat expected when openpilot forces a disengagement of the cruise control. We'd merge a PR to fix this, but it's not something we're going to spend time on fixing in the near future. I'll keep this open in case someone wants to try to fix it.

adeebshihadeh, 2021-12-15:
Moving this to the bounty issue #23224

closed 2021-12-15 by adeebshihadeh