commaai/openpilot #21778: Lane departure warning false positive

grekiki, 2021-07-29:
**Describe the bug**
On the route 650b1e4383a9baf2|2021-07-29--09-50-32 at 210s I got a lane departure alert despite staying clearly inside my lane. 

**How to reproduce or log data**

Not sure it's reproducible, will add info if I get more occurences

**Device/Version information (please complete the following information):**

 - Device:comma2
 - Version: Fork of master from a month ago, very simmilar to 0.8.6 release
 - Car make/model: VW Caddy

Quick bug preview: https://youtu.be/1LKCRRWEi0k

pd0wm, 2021-07-29:
Looks there is quite a bit of yaw to the left after passing the white car. I can imagine the model thinks you're going to cross the line.

grekiki, 2021-07-29:
[image]
Ok the model seems to have quite freqent lane change ideas, looks like at also thought that I am going to lanechange to the right on the sidewalk and then the bus stop afterwards at 50 kph, was "lucky" there were no lanelines close to the driving line. Looking at the rest of the drive it seems to think I am going to perform a right lanechange whenever there is a sideroad to the right no matter my speed...

closed 2021-07-29 by grekiki