commaai/openpilot #21799: Speed

Na112786, 2021-07-30:
**Describe the bug**

**How to reproduce or log data**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

the speed with me it's 111 km / 69 mph.
How I can fix the speed limit to the normal speed limit, 84mph Pls, because its too slow!!

vanillagorillaa, 2021-07-30:
Please provide all information requested in the template.

vanillagorillaa, 2021-07-30:
Closing since looks like you created a new issue #21803

closed 2021-07-30 by vanillagorillaa

Na112786, 2021-07-30:
Device: [comma two]
Dongle ID: [1631748f51929063]
Route: [1631748f51929063|2021-07-28--06-22-45]
Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
Version: [release2 (0.8.6)]
Car make/model: [honda pilot 2019]

Speed limit now 69mph.
Normal speed limit 84mph.
Can you help me to back the speed limit 84mph, Pls.

Na112786, 2021-07-31:
Hello,
It's same problem, only difference I put all ur Requirements.
Please help me
The old ( 21799)
The new (21803)

If there is any missed
please let me know
Thank you