commaai/openpilot #21803: Speed limit problem

Na112786, 2021-07-30:
- Device: [comma two]
 - Dongle ID: [1631748f51929063]
 - Route: [1631748f51929063|2021-07-28--06-22-45]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [release2 (0.8.6)]
 - Car make/model: [honda pilot 2019]

Speed limit problem, 
Everything was great on speed limit 84mph with version 0.8.5
When I updated to version 0.8.6, I found problem with speed limit 69mph. And the speed is low, so please can you help me to back my original speed limit 84 mph, pls.

Milestone: 2000.
Thank you.

pd0wm, 2021-08-02:
Can you be more specific what happened? Did you get the alert about speed to high? Did the car never reach the desired speed?

Looking at the route I never see the set speed set higher than 101 kph (~63 mph).

Na112786, 2021-08-02:
>
> This is ur question :

> Did the car never reach the desired speed?
>
My answer, YES THE CAR NEVER REACH THE DESIRED SPEED.

This is ur comment :

> Looking at the route I never see the set speed set higher than 101 km ( 63
> mph).
>
My answer, myvcomma two version 0.8.5 reach to (84mph), Attached guide to
the picture.  And all the comma. ai members said, the speed limit is
(84mph), Attached guide to the picture.

Now, the specific problem is attached in video, so u can understand
directly, ( I set the speed on 135km, then the speed come down
automatically to 114km)????

Why this happened in version 0.8.6?
Could you help me, Pls.

>

pd0wm, 2021-08-02:
I don't see a video. If you want to share files you have to do that through the github web interface, not through email.

Are you sure you sent the right route, this is a plot of the set speed while openpilot was engaged in the 2nd half of the drive.

[image]

Na112786, 2021-08-02:
Attache video..
Sorry the video can't send because the big size. I will put in the Honda,
accra comma ia members

Na112786, 2021-08-02:
How I can send the video, in GitHub, in which place?

> Attache video..
> Sorry the video can't send because the big size. I will put in the Honda,
> accra comma ia members, but still big size...
>

Na112786, 2021-08-02:
This is the pic of the route...

> Are you sure you sent the right route, this is a plot of the set speed
> while openpilot was engaged in the 2nd half of the device
>

Na112786, 2021-08-03:
I sent the video to ***@***.***

Please, I need help, do understand me.
Now u have all the information

Na112786, 2021-08-03:
Finally ur system except the video 

https://user-images.githubusercontent.com/88205149/128094328-40f19cd4-c264-4531-af47-74436b40ec55.mp4

Na112786, 2021-08-03:
[image]

Na112786, 2021-08-03:
[image]

Na112786, 2021-08-03:
[image]
[image]

Route

HaraldSchafer, 2021-08-04:
Thanks for the detailed feedback, this issue is the same as this. So I will link and close.
https://github.com/commaai/openpilot/issues/21697

closed 2021-08-04 by HaraldSchafer

Na112786, 2021-08-06:
Hello, 
What is the news about this problem 👇
* Poor longitudinal tuning *

HaraldSchafer, 2021-08-12:
https://github.com/commaai/openpilot/commit/a8f3e2b3c26f2b617a1c9c94250b44adce861d04

Just merged a potential fix into master. This should hopefully help with some Hondas. I was not able to test some, so any feedback is welcome.