commaai/openpilot #21816: Comma Two - 2020 Corolla Hybrid European market- "Dashcam Mode Car Unrecognized"

ibrahimyuzu, 2021-07-31:
Installed Openpilot on a new Comma Two that arrived today.

Message onscreen is "Dashcam Mode Car Unrecognized"

I’m out of USA. Living in Turkey. I thing this european cars firmware different from USA markets. 

Cold you help me how to solve this issue?

pd0wm, 2021-08-01:
Please fill out the bug report template. 

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

ibrahimyuzu, 2021-08-01:
Hi there,

I’m trying to use comma2 in Turkey. My car is European car maybe it has different firmware thats why comma2 does not recognize. 

You find information below. 

> Device: [comma two]
> Dongle ID: [c96e8dd60cad469b]
> Route: [c96e8dd60cad469b|2021-08-01--01-03-14--0]
> Timestamp: [its never worked thats reason I can’t give specific time]
> Version: [0.8.6]
> Car make/model: [Toyota Corolla Hybrid 2020 European Market]

Ibrahim Yuzugulec

vanillagorillaa, 2021-08-01:
You need to fill out the requested information in order for someone to look at. 

Please fill out the bug report template.

Device/Version information (please complete the following information):

Device: [e.g. EON/EON Gold/comma two]
Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
Car make/model: [e.g. Toyota Prius 2016]

ibrahimyuzu, 2021-08-01:
Hi there,

I already reply this message to Willem Melching. But, I’m sending again. 

I’m trying to use comma2 in Turkey. My car is European car maybe it has different firmware thats why comma2 does not recognized my car. 

You can find information below. 

> Device: [comma two]
> Dongle ID: [c96e8dd60cad469b]
> Route: [c96e8dd60cad469b|2021-08-01--01-03-14--0]
> Timestamp: [its never worked thats reason I can’t give specific time]
> Version: [0.8.6]
> Car make/model: [Toyota Corolla Hybrid 2020 European Market]

Ibrahim Yuzugulec

dorkmo, 2021-08-01:
Since the firmware might be slightly different on your car, you might want to go ahead and create a new fingerprint. There are some instructions on the wiki, or the folks on discord might be able to help.

pd0wm, 2021-08-02:
Add the missing fingerprints. You have to run `master-ci` or wait for the 0.8.8 release.

closed 2021-08-02 by pd0wm

ibrahimyuzu, 2021-08-02:
Dear Willem,

Do you know how long does it take to release 0.8.8?

Ibrahim Yuzugulec

pd0wm, 2021-08-02:
That might be a few weeks out. Join our discord at discord.comma.ai if you need help running our nightly build (`master-ci`).

ibrahimyuzu, 2021-08-02:
I have succesfully fingerprinting but still didn’t work.

I don’t know what is master-ci. 

Please can somebody help me? Otherwise I need to return device sadly :(

Ibrahim Yuzugulec

pd0wm, 2021-08-02:
Your best bet is to join the #installation-help channel on our discord (https://discord.comma.ai)

ibrahimyuzu, 2021-08-03:
Dear Williem,

I tried to master-ci but still not working. Someone helped me on discord also, he couldn’t successful. 

And now, c2 doesn’t see the car. It says no panda. 

I tried to uninstall open pilot and install from https://openpilot.comma.ai same it says no panda. 

Ibrahim Yuzugulec

ibrahimyuzu, 2021-08-04:
Hi there,

As I mentioned before email, my comma2 didn't work on my 2020 corolla hybrid. I try to get help from discord and couldn't solve my problem. At this point I have 2 issues.

I guess the most immediate issue is that I keep getting No Panda and the system doesn't work.

I tried to flash Panda , https://github.com/commaai/openpilot/wiki/Panda-Flashing  also, this doesn't work as well.

I attched 3 screenshots of Panda flashing and list of USB connections.

Best Regards.

Ibrahım Yuzugulec

Ibrahim Yuzugulec

PR #21824 (2021-08-02): Add 2020 Corolla Hybrid FW versions
Closes #21816 

`c96e8dd60cad469b|2021-08-01--01-03-14`