commaai/openpilot #21837: Changing the steering torque / steering ratio

jbirtel, 2021-08-03:
**Is it possible to change the highest possible steering torque/ratio of the vehicle spawned by openpilot?**

I mean, if the LKA is not able to manage certain tight curves, that is okay, but shouldn't we be able to 'override' the LKA like in reality such that we can intervene if the LKA is not possible to drive through that curve alone?
I am also not able to steer manually enough to master tighter turns such as 90 degree turns with the openpilot vehicle, even though this is generally possible with CARLA. (Apart from the fact that simultaneous straight ahead and right or left driving is not possible, because only one key is recognized in the bridge)

I already tried to comment out the lines [here](https://github.com/commaai/openpilot/blob/d7ab611bb0a616374e28eb4f0cdf46bb0172bb4b/tools/sim/bridge.py#L185), 

And I also tried to increase the steering ratio [here](https://github.com/commaai/openpilot/blob/master/tools/sim/bridge.py#L31)

**Additional context**

I am running on Ubuntu 18 with openpilot version 0.8.4 and I run it in the docker.

adeebshihadeh, 2021-08-05:
Yes, the CARLA bridge uses the Honda Civic. [Here's] where the civic parameters are set. Check out the simulation channel at https://discord.comma.ai for discussion about the simulator.

closed 2021-08-05 by adeebshihadeh

jbirtel, 2021-08-06:
Thank you, I will have a look at those files. But are you sue that it is the Honda Civic, because I can see a Tesla sign at my car :D ?