commaai/openpilot #21839: Palisade Car Type Bug

Palispades, 2021-08-03:
I have a 2021 Palisade SEL with premium and convenience package with a known comma drive bug.  Whenever I get into a complete stop situation with opendrive enabled, opendrive will start spamming cluster in the stop and go situation, which is normal.  What isn't normal in this sequence attached in the video, is all the lights on the tach will flash on, then flash off, including transmission settings etc, not sure if this step could mess with the electronics in the Palisade.

Comma 2 device.

716dc83a1b1e5e83

Route and timestamp will need to be re created on the way back from work today, currently running the newest software 8.6.  Also getting quite a bit of poor GPS signal, is this common on the comma 2 device?

https://user-images.githubusercontent.com/87155950/128033311-352090ed-6a62-4477-a59e-52a9ae698f18.mp4

adeebshihadeh, 2021-08-04:
@gregjhogan have you seen this?

gregjhogan, 2021-08-04:
> @gregjhogan have you seen this?

I have not, but I have a 2020 Limited FWD, so I would be suspicious of it being caused by a trim difference, drive train difference (e.g. AWD), or model year difference.

Palispades, 2021-08-24:
Comma 3
769957055eea1203
769957055eea1203|2021-08-24--07-30-02 
Fault occurs 17 minutes exact into the drive.
Newest drive Version.
2021 Palisade SEL Premium and Convenience AWD
[image]

Let me know if more info is needed. 

Thanks

sshane, 2022-07-07:
Should be largely fixed on master, please re-open if it isn't (if it happens once when resuming, then might need to lower the message rate) https://github.com/commaai/openpilot/pull/24873

closed 2022-07-07 by sshane