commaai/openpilot #21842: BSM not working on 2019 Toyota CH-R

cferra, 2021-08-03:
BSM not working on Toyota CH-R

Toyota interface.py is filtering out no_dsu_cars for BSM.

The following line is in the toyota interface.py

ret.enableBsm = 0x3F6 in fingerprint[0] and candidate in TSS2_CAR
# Detect smartDSU, which intercepts ACC_CMD from the DSU allowing openpilot to send it

by deleting the "and candidate in TSS2_CAR" blind spot monitoring works during lane changes with my CH-R. Speaking with @mlp changing "TSS2_CAR" in that line to "NO_DSU_CAR" would include the no dsu cars.

The additional check for the 0x3F6 fingerprint would disable BSM for cars that might be in the NO_DSU_CAR group but does not have BSM allowing for those that do, to have BSM.

Device/Version information (please complete the following information):

Device: comma two]
Dongle ID: [2876aae3c6df4380
Route: N/A
Timestamp: N/A
Version: 0.8.5
Car make/model: 2019 Toyota CH-R XLE

cferra, 2021-08-04:
Speaking with @mlp changing "TSS2_CAR" in that line to "NO_DSU_CAR" would include the no dsu cars, the additional check for the 0x3F6 fingerprint would disable BSM

This did not work, may be a manual inclusion of CH-R to the list is required?  or my syntax for the edit is wrong..

adeebshihadeh, 2021-09-04:
This isn't really an issue. Many toyota models have `0x3F6`, but they don't all have the blindspot data. However, we know that all the TSS2 cars have BSM. If you'd like to get this added for your C-HR, open a PR and include a route with a blocked lane change due to BSM.

closed 2021-09-04 by adeebshihadeh