commaai/openpilot #21851: Jerky Steering In Banked Corner

Achilles308, 2021-08-04:
**Describe the bug**

In a decently steep downward hill with a left turn, while OP was engaged, the car was smooth for a little bit into the corner and then made a sudden jerky movement in steering. In the past, this turn would cause ping ponging. Now instead of ping ponging it's very jerky at times. It feels like it was making the right moves, but needs a little more ramp up time on the move for this sweeping style of turn, instead of so suddenly. OP did not behave this way Friday and I first noticed this behavior on Monday. There are other segments of jerky as well on Monday but I can't find the timestamps just yet.

**How to reproduce or log data**

It's happened four times in like 3 hours of driving (maybe an hour 15 minutes of open pilot engaged), so pretty rare. I will say, the 4th time was a pretty well banked tight turn, and it helped OP make the turn better, so it seems like the radius of turn might impact the usefulness of the jerkiness.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f39975d916ed458a
 - Route: f39975d916ed458a/2021-08-03--07-27-19/34
 - Timestamp: Segment 34, the local time was 8:01::34 not sure the exact timestamp
 - Version: Not sure the commit, I hope it is in the logs, but master-ci 0.8.7
 - Car make/model: 2020 Toyota Corolla Hatchback

**Additional context**

I don't believe it was windy. I did notice on replay that maybe it had to do with the bumps or concrete expansion joints in the road? Hard to say. I can look for the other one's if there is something to it, let me know. Tomorrow should bring more as well.

ClockeNessMnstr, 2021-08-04:
Probably different but had some oversteer/correction on long tighter turns 

Sounds sort of similar

Achilles308, 2021-08-05:
It did better today on that turn. See f39975d916ed458a/2021-08-04--06-52-25/18/ around local time of 7:10::36 it hits that same bump. The only difference was time of day, a full tank of gas, different lead car, and master-ci 0.8.8.

I do think it has something to do with the large bumps in the road. I had it happen twice in a row on large bumps from a bridge. See f39975d916ed458a/2021-08-04--06-52-25/14/ at 7:06::49 local time. At the white concrete part of the bridge, the start and ending caused jerky steering due to the bumps.

In general, the turning is improving even if it is SOMETIMES a little jerky. Today on the way home, it was turning even better with no noticeable jerks. Planning was a little close to the center at times but otherwise great stuff.

Side note, I tried to take a traffic circle and it just wouldn't steer far enough to make it. Had OP slowed down, it likely would have. Might try tomorrow if traffic allows such and manually force it to slowdown.

Achilles308, 2021-08-05:
> Probably different but had some oversteer/correction on long tighter turns
> 
> Sounds sort of similar
> 
> #21346

Interesting you call it oversteer... when my car hit the bump yesterday, it caused oversteer and then it jerked into the corner to correct. It was jarring maybe because it turned the wheel while it was light in the air? So when it hooked up, it was more violent than normal steering. I was thinking if I was driving and did the same jerk, would it feel normal? I think so, because you would anticipate it. So, OP is too good to make moves outside of it's normal behavior, if that makes any sense.

I'll see tomorrow if anything improves :)

Achilles308, 2021-08-10:
It hasn't done the jerky motion over a bump for the last 3 days... so if that continues I think we can close this one. Seems like maybe it was a car tune issue?

Also, turning in sharp corners is much improved. Liking the direction it is going 👍

Achilles308, 2021-08-18:
This hasn't been an issue since like 4 days after the initial bug report. The tune has been fixed. Liking the new torque and smoothness to it as well!

closed 2021-08-18 by Achilles308