commaai/openpilot #21873: Openpilot fails to activate with error "Communication Issue between Processes:

trentrand, 2021-08-07:
**Describe the bug**

For the past week I have been unable to use Openpilot. Suddenly an error "Communication Issue between Processes" is displayed every time I attempt to activate openpilot.

**How to reproduce or log data**

This issue can be easily reproduced by simply activating openpilot at any time.
The issue occurs 100% of the time.

**Device/Version information:**
 - Device: comma two
 - Dongle ID: 27858928f280bac1
 - Route: 27858928f280bac1|2021-08-04--20-45-36
 - Timestamp: 20:25:40, about 5 seconds into the drive. Occurs at every green blip in the route timeline.
 - Version: 0.8.7
 - Car make/model: Toyota Rav4 2021

**Additional context**

I have attempted to find solutions on line, and performed some steps in hopes of resolving the issue:
- Unplugged the Panda and its ethernet on both ends
- Unplugged the LKAS/LDA cable from the car harness 
- Unplugged the USB-C cable from the car harness to the Comma Two
- Plugged them all back in

This didn't work, so I also ensured that the Comma Two was working correctly:
- Uninstalled and re-installed OpenPilot 0.8.7
- Ensured the driver-facing camera was operating by viewing camera feed

pd0wm, 2021-08-08:
Can you upload the raw logs? Turn on your device, go to https://my.comma.ai/useradmin/?onebox=27858928f280bac1%7C2021-08-04--20-45-36 and click "upload all logs".

trentrand, 2021-08-09:
Done! Everything should be uploaded.

pd0wm, 2021-08-09:
I don't see any traffic on the radar bus. This probably means there is a wiring or connection issue. Try reseating *all* the connectors. Sometimes flipping around the usb-c cable on one side helps as well.

trentrand, 2021-08-12:
Good news! I tried again by unplugging the panda and Ethernet, as well as the car adapter and its connections to the radar & lkas camera, Ethernet and usb-c to Comma Two.

The device began functioning correctly after everything was reattached!

Thanks @pd0wm !

closed 2021-08-12 by trentrand

trentrand, 2021-08-18:
My device continued to have this error after the next time, and other times an error of 'car unrecognized' was displayed.

I've swapped the USB-C cable from the car harness to comma two and the issue went away.

Euclid71, 2023-06-10:
I get this error with the same symptoms when I have the OpenStreetMap database set to "== None ==".  When I set the database to "US - Pacific", I never get this error.  

Route: f1bc09b7aead608d|2023-06-09--23-37-12--1 
All logs are uploaded.