commaai/openpilot #21882: No SNG support  2021 Camry

OPgood, 2021-08-09:
No SNG support on most car models without Full-Speed Range DCC (basically any car from Toyota that does not have a electronic brake from 2021+) how can this be fixed? Specifically on the 2021 Camry.

adeebshihadeh, 2021-08-09:
This is not a bug, as this is the behavior of your stock ACC system. However, you may be able to workaround it, and we'd merge a PR to do so.

closed 2021-08-09 by adeebshihadeh

OPgood, 2021-08-09:
So it’s that’s the case they brew to fix the website! Website clams it works at 0 acc

VirtuallyChris, 2021-08-09:
Where on the website does it claim it works stop and go?

OPgood, 2021-08-09:
Did you not just read my previous comment? Website (this GitHub itself compatibility chart) clams it works to 0ACC not 20ACC so if it was 0ACC like it states it would work for stop and go not automatically disengage at 20mph

OPgood, 2021-08-09:
[image]

dorkmo, 2021-08-10:
looks like the footnote reference [4] talks about some camrys not having full speed range dynamic radar cruise control, not sure if it is possible to get around this. You might ask on the toyota discord if anyone has tried

dorkmo, 2021-08-10:
@VirtuallyChris would it be better to list it as 28mph, then have the reference talk about 0mph on certain trims?
Does pedal work in this scenario?

anotabien, 2021-09-06:
@AlexandreSato was working on a workaround to enable SNG
https://github.com/AlexandreSato/openpilot-Brazil/blob/master/acc_type1.py
https://medium.com/@tapecom/openpilot-stop-and-go-hack-for-thailand-toyota-corolla-cross-hybrid-premium-sefty-model-with-comma-b379a2a95bbc