commaai/openpilot #21887: Braking For Turn Lane Cars

Achilles308, 2021-08-10:
**Describe the bug**

Incident one:

A car was waaaaaay in front of me turning right from the right turn lane. OP broke like the car wasn't in the turn lane. To be fair, the car took a long time to turn, and was stopped by the time they turned. So radar must have been slightly confused. I initially thought OP was braking for the car that turned in front of me, but really neither were cause for braking at that slow of a speed.

**How to reproduce or log data**

The same behavior happened twice in this drive.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f39975d916ed458a
 - Route: f39975d916ed458a|2021-08-09--18-01-37--2
 - Timestamp: 18:04::08 local time is roughly when it starts through that light
 - Version: master-ci 0.8.8 (had just updated before start of drive)
 - Car make/model: 2020 Toyota Corolla Hatchback

**Additional context**

I'm pretty sure this incident and the other one I posted below are the same bug.

Achilles308, 2021-08-10:
**Describe the bug**

Incident two:

Was coming up over a small hill and there was a stopped car in the left turn lane. OP started to brake like maybe it thought it was in my lane. I jumped on the throttle and continued on.

**How to reproduce or log data**

The same behavior happened twice in this drive.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f39975d916ed458a
 - Route: f39975d916ed458a|2021-08-09--18-10-48
 - Timestamp: 18:39::35 local time is roughly when it starts through the car in the left turn lane
 - Version: master-ci 0.8.8 (had just updated before start of drive)
 - Car make/model: 2020 Toyota Corolla Hatchback

**Additional context**

OP likely would have broke harder, but I caught it and accelerated through and re-engaged shortly after.

Achilles308, 2021-08-10:
Also, watching the high resolution video makes it clear these are likely some kind of car crossing lane edge case where it isn't sure which lane the stopping or stopped car is in. They are a lot more edge case than I thought both live and even in the low resolution versions. Hope it helps!

HaraldSchafer, 2021-08-10:
We're in the process of reworking a lot of the longitudinal stuff, I'll keep this in mind and hopefully this kind of stuff will be improved.

Achilles308, 2021-08-10:
> We're in the process of reworking a lot of the longitudinal stuff, I'll keep this in mind and hopefully this kind of stuff will be improved.

I did not notice any further instances of this on my drive home... not sure the commit though or if I updated it at some point. How can I pull the commit version from the logs?

Achilles308, 2021-08-24:
It has happened again. Several of the lead cars made it into the left turn lane and were either almost stopped or were stopped. I was still in the lane of travel and OP started to brake. It was in a moderate right turn which means the front of my car was pointed at the cars turning left in the turn lane.

comma 3
master-ci 0.8.8 (had the most recent updated right before the start of drive)
dfb993cc43ba69dc|2021-08-23--19-00-18--29
19:29::38 local time is roughly when it starts
2020 Toyota Corolla Hatchback

I had to jump on the throttle disengaging OP and continue on my way.

Achilles308, 2021-09-02:
Pretty bad example of OP braking for a car in the turn lane again. The car was clearly in the marked turn lane and broke from 35 to 30 mph before continuing on. In this case, the lead car used their turn signal. The cut out detection was abysmal (I know cut out is not a real thing, but it likely should be!)

comma 3
master-ci 0.8.9 (had the most recent updated right before the start of drive)
dfb993cc43ba69dc\|2021-09-01--18-07-32--44
18:52::12 local time is roughly when it starts
2020 Toyota Corolla Hatchback

The white object in the road after the turn lane incident turned out the be an empty plastic bag... but I swerved to avoid it all together.

closed 2021-09-13 by adeebshihadeh