commaai/openpilot #21888: Yellow Line Hugging At No Line Section

Achilles308, 2021-08-10:
**Describe the bug**

There is a section of road that I daily drive where it always hugs very aggressive towards the double yellow line (full line mode). This is wider section of the road than the previous parts and the lines that were very consistent just suddenly stop for a few feet. Something about this configuration really causes OP to hug hard left near the yellow line (nearly touching it), despite the lane being quite wide.

**How to reproduce or log data**

Happens every time here.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f39975d916ed458a
 - Route: f39975d916ed458a\2021-08-09--18-10-48--47
 - Timestamp: 18:58::23 local time give or take
 - Version: master-ci 0.8.8
 - Car make/model: 2020 Toyota Corolla Hatchback

**Additional context**

I suspect the 'don't take the exit' part of the model is being triggered here and is forcing the car further into the lane than needed (even if it was an exit!). I could try lane less here if that contrast would help?

Achilles308, 2021-08-10:
I've noticed this happen a few more times. I think on laneful model this exact scenario is a major contributor to lane hugging on surface roads. It'll be good and centered until the gap in the double yellow and the gap in the right white line, then it hugs the yellow. If I can find the other examples, I'll post them.

Achilles308, 2021-08-11:
@HaraldSchafer I found the one that really stuck with me... OP was doing so great on this road until the hill and gap in lines. Maybe it is a hill problem on this one and not a line gap problem, but it was pretty aggressively close to the double yellow line.

Device: comma two
Dongle ID: f39975d916ed458a
Route: f39975d916ed458a|2021-08-10--12-30-24--3
Timestamp: 12:33::53 local time
Version: master-ci 0.8.8
Car make/model: 2020 Toyota Corolla Hatchback

It is a similar road as well as the first one I posted, being a right turn lane with gaps in all lines at that section.

Achilles308, 2021-08-11:
> There is no 'don't take exit' part of the model. I appears to be a tiny bit right in the lane, but not that much. Really it should react to the oncoming traffic and give it space. But only the laneful model can never do that.
> 

I got a closer screen grab to when the line ended from the high resolution video. About at your screen grab it started to turn and was too aggressive...  So it definitely wasn't in the right side of the lane. Look at how worried the truck was in getting over in his lane. I'm not sure if or when I took over, the logs will make that clear, but it is a very wide road and struggles with the gap in lines on laneful mode.

[image]

HaraldSchafer, 2021-09-04:
Closing this, this is part of the general improving laneless driving project. Not really actionable with a specific fix, thanks for the report!

closed 2021-09-04 by HaraldSchafer

Achilles308, 2021-09-23:
> Closing this, this is part of the general improving laneless driving project. Not really actionable with a specific fix, thanks for the report!

Yesterday, this part of my drive was perfect at not hugging! This was laneful.

comma 3
master-ci 0.8.10
dfb993cc43ba69dc|2021-09-22--18-41-11--43
19:24::50 Local time
2020 Toyota Corolla Hatchback

Really good update and look forward to this further improvements from here!