commaai/openpilot #21889: Openpilot repeatedly crossing lane lines on a specific set of road

SushiFerret, 2021-08-10:
**Describe the bug**

On a specific stretch of road, Openpilot will make a clear cut across the lane lines as if changing lanes but with no indication. This has shown to be a repeatable event as well, while the lane markings here are admittedly very poor, it seems to have no problem driving similarly marked roads. But for some reason, this specific stretch of road with these specific lane markings seem to consistently confuse it and cause it to cross over into the next lane. 

**How to reproduce or log data**

All that is needed is to drive over this specific stretch of road normally with Openpilot enabled. 

**Device/Version information (please complete the following information):**
 - Device: Comma Two
 - Dongle ID: 6e8eb47ac4dcfd94
 - Route: 6e8eb47ac4dcfd94|2021-08-07--17-58-51
 - Timestamp: 18:24:52
 - Version: release2 0.8.7
 - Car make/model: 2018 Toyota Camry SE 

**Additional context**

https://user-images.githubusercontent.com/65090064/128800766-05035a76-9822-41e9-900c-6e098523358f.mp4

Achilles308, 2021-08-10:
We were discussing this one in the openpilot-experience discord.

This particular video reminds me of a merge that I have to take. I noted that if you look at the kind of black lines that are diagonal to the lanes (I'm assuming these are old tar pads where the reflectors were stuck down... either that or reflectors in full shadow) that it looks a lot like this merge f39975d916ed458a|2021-08-09--06-45-02--12 starts at around 6:57::27.

In this merge, the lane I'm in has a solid white left and solid white right line and turns into a white dashed left line and a solid right line. The dash section indicates the area you are permitted to merge into. OP has always hugged that right line until near the very end where it will jump the dashed line and merge left then center in the new lane. It seems to me, that is what happened in this instance as well, except the black lines were not a new lane, but unrelated fluff to the actual lanes. It actually stayed inside the lane when the reflectors were present and didn't seem to merge over until the reflectors were missing in a large section.

Also, what would the car have done on SushiFerrets drive if a car was beside and the BSM was indicating a car was there? In other words, was this treated as a lane change and would have been blocked due to the car being in the blind spot? Or did it just think that was the same lane and would have driven into the side a car beside it?

Anyways, hope this helps squash this annoying bug!

HaraldSchafer, 2021-08-10:
This failure doesn't surprise me at all. It took my several views of the video to even understand what is happening. There is a laneline on the right that the model is following that has poorly been painted over, which diverges from the botts' dots. While the "actual" lane indicated by the botts dots goes right, into the wall. Honestly some of the worst road markings I've ever seen.

@Achilles308 I looked into your failure too, it doesn't look particularly smooth. But if you didn't intervene it handled it okayish?

Achilles308, 2021-08-10:
@HaraldSchafer I agree it is poor lane markings. It appears to be a construction zone, and lane shifts like that are not that uncommon.

The merge that I take has better days and okay days. In general, OP on the worst day is better than the worst driver I've seen take it. OP sometimes touches the solid white right line, but never crosses it towards the wall. It could be smoother in merging in earlier kind of centered in the opening but it really isn't a concern over other issues. Overall it does it okay I'd say.

**Also, what would the car have done on SushiFerrets drive if there was a car was beside and the BSM was indicating a car was there? In other words, was this treated as a lane change and would have been blocked due to the car being in the blind spot? Or did it just think that was the same lane and would have driven into the side a car beside it?**

HaraldSchafer, 2021-08-11:
Openpilot thought it was following the lane, so no, it would not have been blocked.

closed 2021-08-11 by HaraldSchafer