commaai/openpilot #21893: OpenPilot not slowing down to take highway turns in Carla

JackWetherell, 2021-08-10:
**Describe the bug**

When openpilot attempts to take turns at a given high set speed on the highway in the CARLA simulator, it doesn't not adjust the speed in order to take turns safely, and often crashes if the target speed is set too high.

Is this expected behaviour?

**How to reproduce or log data**

Run openpilot in CARLA as described in the README, and press 1 many times to set a high speed (~50 MPH), and it will crash due to not slowing down to take a turn. 

Operating system: Ubuntu 20.04.

pd0wm, 2021-08-10:
This is expected behavior. Openpilot does not slow down for turns.

closed 2021-08-10 by pd0wm

JackWetherell, 2021-08-10:
Thanks for the info @pd0wm. Is there a simple explanation why that is, as It slows down for following the leading vehicle. Many thanks.