commaai/openpilot #21896: Unprompted Exit

Achilles308, 2021-08-10:
**Describe the bug**

With OP engaged, unprompted, it started to turn the car into taking the exit. I let it go hoping it would correct its error, but alas, it wasn't to be. I took over the wheel and brought it back into the lane OP should have stayed in.

**How to reproduce or log data**

I've had issues at this exit off and on in the past. I can't recall which laneful or laneless did what though or the versions at the time.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f39975d916ed458a
 - Route: f39975d916ed458a|2021-08-10--07-05-15--41
 - Timestamp: 7:46::58 local time is about where it starts
 - Version: master-ci 0.8.8
 - Car make/model: 2020 Toyota Corolla Hatchback

**Additional context**

This particular part of the road has these deep grooves in the road on the left and a seam where the recently paved right size meets up with the older left side. Lighting was pretty poor as the sun was directly over the trees facing that eastern direction. The lead car signaled for 1/2 a mile and slowly took the turn, which maybe factored into the follow.

Achilles308, 2021-08-12:
I am happy to report, at the same exit today, OP did not take the exit! 

f39975d916ed458a|2021-08-11--06-51-19--43
7:34::42 local time
comma 2 on master-ci 0.8.8
2020 Toyota Corolla Hatchback

Additional context

I had just finished an automatic lane change. It hugged a little bit towards the exit (maybe a full tire width shifted over) and then stopped turning and just continued on pretty straight.

Achilles308, 2021-08-13:
I was not able to hit that exit today. On a different exit today, it hugged and thought about going but didn't. This exit was extremely well marked and it still thought about taking it.

Device: comma two
Dongle ID: f39975d916ed458a
Route: f39975d916ed458a|2021-08-12--06-46-02--24
Timestamp: 7:10::50 local time
Version: master-ci 0.8.8
Car make/model: 2020 Toyota Corolla Hatchback

Of note, I took some automatic lane changes in the sharp curves before the exit... and it struggled a little bit but overall wasn't bad.

HaraldSchafer, 2021-08-16:
New model has had some fixes to the exit behavior, should be big improvement! Lemme know if it worked.

closed 2021-08-16 by HaraldSchafer

Achilles308, 2021-08-17:
I'll try laneful tomorrow, but today at that troubled exit it took the exit again on laneless (I updated the master-ci right before the drive).

To be fair, there was a lot going on with a cut in, it's a two lane exit with a car taking the far right lane exit. It's probably an edge case. I had a car seemingly passing me on the right as they took the exit, so I couldn't let it go into that lane and took over pretty quickly. It might have caught that line at the last second and corrected but hard to say. I can switch between laneless and laneful if it would help?

comma 2
f39975d916ed458a|2021-08-16--09-09-32--5
9:14::45 local time the start of the exit process
2020 Toyota Corolla Hatchback

Achilles308, 2021-08-18:
> New model has had some fixes to the exit behavior, should be big improvement! Lemme know if it worked.

Took this exit in the POURING rain from Tropical Depression Fred... OP did it PERFECT on laneful!

comma 2
f39975d916ed458a|2021-08-17--06-59-58--50
7:50::39 local time the start of the exit process
2020 Toyota Corolla Hatchback

Looking at the high resolution, I think it was following the tire tracks in the rain. But, either way, it took this exit solid.

HaraldSchafer, 2021-08-18:
Great!

Achilles308, 2021-08-19:
> Great!

Even better one on Wednesday. So truly and utterly solved, it's like magic.

comma 2
master-ci 0.8.8
2020 Toyota Corolla Hatchback
f39975d916ed458a|2021-08-18--06-57-39--42
07:39::45 local time start of the exit

Achilles308, 2021-08-20:
> Great!

Unfortunately, on laneless, it took the exit =/ 

comma 2
master-ci 0.8.8 (updated right before the drive)
2020 Toyota Corolla Hatchback
08:35::00 Local time
f39975d916ed458a|2021-08-19--07-49-33--45

I also had a poor 'exit' take with laneless. Basically, it cut way to close to that other entrance and guardrail.

Same drive as above...
f39975d916ed458a|2021-08-19--07-49-33--29 and 30
08:19::30 local time