commaai/openpilot #21909: Vehicle Parameter Identification Fault

sjburt, 2021-08-11:
**Describe the bug**

While driving, got red alarm with "Vehicle Parameter Identification Failed" and disengage. After that, cannot re-engage.

**How to reproduce or log data**

Happens intermittently, maybe once every week or two. Goes away after car is stopped and started. Cannot reproduce

https://connect.comma.ai/457d42782c83b77c/1628687689031/1628689942362

**Device/Version information (please complete the following information):**
 - Device: commatwo
 - Dongle ID: 457d42782c83b77c
 - Route: 457d42782c83b77c|2021-08-11--06-14-49
 - Timestamp: At clock time 6:29:50
 - Version: Release 0.8.7
 - Car make/model: Toyota Rav4 Hybrid 2021

**Additional context**
None, doesn't seem associated with anything that I can pinpoint.

HaraldSchafer, 2021-08-12:
@nuwandavek Can you check this out?

nuwandavek, 2021-08-12:
@sjburt, we investigated this and we suspect the issue is in the wheel alignment of your car. This alert is triggered when the offset angle (the steady state steering angle required to make the car go straight; this accounts for steering wheel errors, road bank etc.) crosses a certain threshold. So, as you can see in the plot, despite going straight, the steering wheel is ~10 degrees off of its zero position, which is outside our allowed threshold. 

[image]

sjburt, 2021-08-12:
Interesting, thank you. I wonder why the issue occurs occasionally and why it would occur on a new car. I will check some other drives and see if it is a consistent angle near the threshold...

sjburt, 2021-08-12:
Ok, I took a look at 3 more drives and it is quite confusing. Sometimes it is fine. Sometimes it steadily goes out of alignment. What would cause this? It seems like a mechanical alignment issue would result in a consistent angle in the same direction.

This is the drive I first posted, 457d42782c83b77c|2021-08-11--06-14-49
[image]

However, on the next drive (457d42782c83b77c\|2021-08-11--08-43-33) the steering angle seems to be fine.
[image]

Here are two more drives from a previous day. Same route in opposite directions. 
457d42782c83b77c|2021-08-10--17-18-41
[image]
457d42782c83b77c\|2021-08-10--18-46-37
[image]

nuwandavek, 2021-08-12:
Yeah, even I was looking into routes before and after, and was confused. I'm looking into this now and will get back to you.

nuwandavek, 2021-08-12:
@sjburt Could you upload all the rlogs for this particular route?

sjburt, 2021-08-12:
Will do, might take a few hours or overnight, it seems to have some old logs & videos it wants to upload first. I will update the ticket when everything is done.

sjburt, 2021-08-13:
Rlogs are now uploaded.

pd0wm, 2021-08-15:
Quick update on this. Most TSS2 Toyotas have two angle sensors. One in the `STEER_ANGLE_SENSOR` message, and one in the `STEER_TORQUE_SENSOR` message. The `STEER_TORQUE_SENSOR` one is more accurate, but it always starts at 0 on car start independent on steering wheel position. Therefore we compute the offset between `STEER_ANGLE_SENSOR` and `STEER_TORQUE_SENSOR`  sensors on car startup and then use the accurate sensor with the offset added.

However, on your car it seems like the two angles are slowly drifting apart and the offset is not constant (see plot below). This forces the angle offset learner to learn this offset until it triggers the sanity check. My guess is that it's an issue with the Rav4s with the alternate EPS (the FW starting with `\x02`, see https://github.com/commaai/openpilot/pull/21429#issuecomment-873652891).

The solution will be to keep updating the offset while driving. I'll open a PR for this, and we'll also see what cars are exactly suffering from this issue.

[image]

closed 2021-09-01 by pd0wm

NylonDiamond, 2024-06-17:
@sjburt @nuwandavek I think this may be the same issue I'm having. How to I get the same chart as the third comment above? Cabana?

PR #21943 (2021-09-01): Continuously update offset between TSS2 angle sensors
Closes #21909 

TODO: 
- [x] ~Figure which cars exactly suffer from this~. edit: the signal is changing too slowly to efficiently search for this. For now assume it's just Rav4s and the Mirai.
- [x] https://github.com/commaai/openpilot/pull/21953