commaai/openpilot #21916: DM False Negative?

Achilles308, 2021-08-13:
**Describe the bug**

I'm not positive, but it seems to me that I received no driver monitoring alerts or warnings when I probably should have. DM camera was off, so no footage I'm afraid. But I've uploaded the logs and front video and hopefully that can help? The road was practically empty, so likely the most lax DM possible. I was looking off towards the left for probably in total 10 to 15 seconds. I was quickly glancing back at the road maybe every 4 seconds, so maybe that reset some kind of timer.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: f39975d916ed458a
 - Route: f39975d916ed458a|2021-08-12--06-46-02--22 and 23
 - Timestamp: 7:08::28 local time (it happened somewhere up or down that hill, like within that next minute and 10 seconds)
 - Version: master-ci 0.8.8
 - Car make/model: 2020 Toyota Corolla Hatchback

**Additional context**

I'd be very curious if there were any alerts or not. If not, then that is a major problem as I totally deserved some alerts.

adeebshihadeh, 2021-08-13:
We can't really look into this without the DM video. Feel free to reopen if you have a route with video.

closed 2021-08-13 by adeebshihadeh

Achilles308, 2021-08-13:
Can you verify if there were alerts?

adeebshihadeh, 2021-08-13:
There were no events in those two segments.