commaai/openpilot #21928: Loss of C3 NVME with kernel panic

jyoung8607, 2021-08-14:
**Describe the bug**

My rev-H production/retail C3 CrossCountry has lost its NVME at least twice.

The first time, the only noticeable symptom was a NVME not mounted alert in the UI. Examination of kernel logs show the NVME was lost within about 30 seconds of being mounted.

The second time, it was lost while onroad and engaged. The screen froze briefly, I had a Lane Assist fault, and the device rebooted on its own. Examination of kernel logs show the NVME PCIe device was lost, and a filesystem-related panic followed.

**How to reproduce or log data**

Unknown. This has never happened with my prior two units.

**Device/Version information (please complete the following information):**
 - Device: Comma Three CrossCountry
 - Dongle ID: 7175cfb6b545809b
 - Route: 7175cfb6b545809b|2021-08-13--08-33-10
 - Timestamp: Not logged; route will have been truncated with loss of NVME and kernel panic
 - Version: jyoung8607/openpilot@f306e470b055bbab1af6e457fcbde41b702d1424 (3616602cd20e328f70d3dd206067e1bd2e78c239 with prototype VW openpilot longitudinal)
 - Car make/model: 2018 Volkswagen Golf R

**Additional context**

If desired, I can open it up, inspect and reseat the NVME. Absent other guidance, I don't intend to open it.

jyoung8607, 2021-08-14:
Relevant logs from first incident a few days ago. Raw logs available if needed.

```
Aug  9 08:32:34 tici kernel: [   41.161645] handle_linkdown_irq: PCIe link is down for RC1
Aug  9 08:32:34 tici kernel: [   42.008180] nvme 0001:01:00.0: Failed status: 0xffffffff, reset controller.
Aug  9 08:32:34 tici kernel: [   42.008547] nvme 0001:01:00.0: of_irq_parse_pci() failed with rc=134
Aug  9 08:32:34 tici kernel: [   42.048059] VFS: busy inodes on changed media or resized disk nvme0n1
Aug  9 08:32:34 tici kernel: [   42.048720] blk_update_request: I/O error, dev nvme0n1, sector 0
```

jyoung8607, 2021-08-14:
Relevant logs from today's events. This was also uploaded in `7175cfb6b545809b|2021-08-13--08-38-07--BOOT`.

```
<3>[  306.381146] handle_linkdown_irq: PCIe link is down for RC1
<4>[  307.063915] nvme 0001:01:00.0: Failed status: 0xffffffff, reset controller.
<3>[  307.064235] nvme 0001:01:00.0: of_irq_parse_pci() failed with rc=134
<4>[  307.084080] pci_raw_set_power_state: 13 callbacks suppressed
<6>[  307.084091] nvme 0001:01:00.0: Refused to change power state, currently in D3
<4>[  307.084153] nvme nvme0: Removing after probe failure status: -19
<6>[  307.084248] nvme0n1: detected capacity change from 1000204886016 to 0
<3>[  308.027082] CAM_ERR: CAM-CDM: cam_hw_cdm_submit_gen_irq: 364 cdm test remove bl tag 3 old ctx ffffff887360f1a8 cookie 4372 new ctx ffffff88736139e8 
cookie 4385
<4>[  308.550127] VFS: busy inodes on changed media or resized disk nvme0n1
<3>[  308.550238] blk_update_request: I/O error, dev nvme0n1, sector 718404880
<4>[  308.550268] EXT4-fs warning (device nvme0n1): ext4_end_bio:314: I/O error -5 writing to inode 22413351 (offset 41558016 size 385024 starting block 8
9800704)
<3>[  308.550280] Buffer I/O error on device nvme0n1, logical block 89800610
<3>[  308.550294] Buffer I/O error on device nvme0n1, logical block 89800611
<3>[  308.550302] Buffer I/O error on device nvme0n1, logical block 89800612
<3>[  308.550310] Buffer I/O error on device nvme0n1, logical block 89800613
<3>[  308.550317] Buffer I/O error on device nvme0n1, logical block 89800614
<3>[  308.550325] Buffer I/O error on device nvme0n1, logical block 89800615
<3>[  308.550332] Buffer I/O error on device nvme0n1, logical block 89800616
<3>[  308.550340] Buffer I/O error on device nvme0n1, logical block 89800617
<3>[  308.550347] Buffer I/O error on device nvme0n1, logical block 89800618
<3>[  308.550350] blk_update_request: I/O error, dev nvme0n1, sector 718413824
<3>[  308.550358] Buffer I/O error on device nvme0n1, logical block 89800619
<4>[  308.550362] EXT4-fs warning (device nvme0n1): ext4_end_bio:314: I/O error -5 writing to inode 22413351 (offset 41943040 size 5677056 starting block 
89801984)
<1>[  308.550823] Unable to handle kernel NULL pointer dereference at virtual address 00000008
<1>[  308.550837] pgd = ffffff8873903000
<1>[  308.550847] [00000008] *pgd=0000000176de8003, *pud=0000000176de8003, *pmd=0000000000000000
<4>[  308.550911] ------------[ cut here ]------------
<2>[  308.550916] Kernel BUG at ffffff887093e484 [verbose debug info unavailable]
<0>[  308.550922] Internal error: Oops: 96000005 [#1] PREEMPT SMP
<4>[  308.550927] Modules linked in: wlan(CE) snd_soc_sdm845(E) snd_soc_wcd9xxx(E)
<4>[  308.550946] CPU: 3 PID: 39034 Comm: kworker/u16:9 Tainted: G         C  E   4.9.103+ #1
<4>[  308.550952] Hardware name: Qualcomm Technologies, Inc. sda845 v2.1 TurboX-SOM_V01 (DT)
<4>[  308.550971] Workqueue: writeback wb_workfn (flush-259:0)
<4>[  308.550981] task: fffffff666394600 task.stack: fffffff6556bc000
<4>[  308.550992] PC is at jbd2_journal_add_journal_head+0x15c/0x218
<3>[  308.550995] blk_update_request: I/O error, dev nvme0n1, sector 718405632
<4>[  308.551001] LR is at jbd2_journal_add_journal_head+0x60/0x218
<4>[  308.551006] EXT4-fs warning (device nvme0n1): ext4_end_bio:314: I/O error -5 writing to inode 22413351 (offset 41943040 size 5677056 starting block 89800960)
<4>[  308.551011] pc : [<ffffff887093e484>] lr : [<ffffff887093e388>] pstate: 60c00145
<4>[  308.551016] sp : fffffff6556bf5c0
<4>[  308.551020] x29: fffffff6556bf5c0 x28: 00000000a1e1e958 
<4>[  308.551030] x27: ffffff887187eb20 x26: 0000000000000000 
<4>[  308.551041] x25: fffffff6720b7000 x24: fffffff661165690 
<4>[  308.551050] x23: 0000000002408040 x22: ffffff887187f670 
<4>[  308.551059] x21: ffffff8872d3edd8 x20: 0000000000000000 
<4>[  308.551068] x19: fffffff5a1e1e958 x18: 0000000000003df1 
<4>[  308.551076] x17: 0000005591cbde90 x16: 0000007f7b649660 
<4>[  308.551086] x15: 0000000000000000 x14: 0000000000000000 
<4>[  308.551095] x13: 0000000000000000 x12: 0000000000000000 
<4>[  308.551106] x11: 0000000000000000 x10: 0000000000000000 
<4>[  308.551115] x9 : 0000000000003ff0 x8 : 0000000000003fff 
<4>[  308.551123] x7 : fffffff67dc84090 x6 : fffffff67dc84090 
<4>[  308.551132] x5 : 0000000000000000 x4 : 0000000001000000 
<4>[  308.551140] x3 : 0000000000000018 x2 : 000000000d62c021 
<4>[  308.551149] x1 : 0000000000000000 x0 : 000000000d62c021 
<4>[  308.551158] 
<4>[  308.551158] PC: 0xffffff887093e444:
<4>[  308.551163] e444  52800300 aa1303e1 940479ad 34fff9e0 d503201f 52800020 97f695af d5384100
<4>[  308.551190] e464  b9401001 34000621 d503201f d503203f f9400260 37c7ffc0 17ffffbe f9402261
<4>[  308.551215] e484  b9400820 11000400 b9000820 d5033bbf 52800300 aa1303e1 94047961 52800020
<4>[  308.551241] e4a4  97f6959d d5384100 b9401001 34000341 b40000b4 90012000 aa1403e1 f9471000
<4>[  308.551267] 
<4>[  308.551267] LR: 0xffffff887093e348:
<4>[  308.551272] e348  910662d6 72a04817 aa0003f3 aa1e03e0 d503201f 90012015 913762b5 d503201f
[... 131 lines cut ...]
<4>[  308.571856] f460: fffffff5a1e1e958 fffffff666f70000 0000000005580004 0000000005580004
<4>[  308.571860] f480: fffffff666f700e8 c0e6aaf5b87e4000 000000000d62c021 0000000000000000
<4>[  308.571864] f4a0: 000000000d62c021 0000000000000018 0000000001000000 0000000000000000
<4>[  308.571867] f4c0: fffffff67dc84090 fffffff67dc84090 0000000000003fff 0000000000003ff0
[... cut ...]
<4>[  308.571928] [<ffffff8870873e28>] wb_writeback+0x108/0x3f8
<4>[  308.571932] [<ffffff8870875008>] wb_workfn+0x108/0x480
<4>[  308.571937] [<ffffff88706cb6bc>] process_one_work+0x20c/0x4d8
<4>[  308.571941] [<ffffff88706cb9d8>] worker_thread+0x50/0x4d0
<4>[  308.571945] [<ffffff88706d2980>] kthread+0x100/0x108
<4>[  308.571949] [<ffffff8870683f00>] ret_from_fork+0x10/0x50
<0>[  308.571955] Code: d503201f f942e660 f9400bf3 a8c27bfd (f85d8000) 
<4>[  308.571960] ---[ end trace 06eb0b4da30615c7 ]---
```

RedBaron2, 2021-08-16:
@jyoung8607 
I know it doesn't happen often, but the log points to sectors of the nvme as being inaccessible. This might mean a failing nvme drive.
Since c3 is using Ubuntu at its core. The disks utility should be able to confirm or deny this by looking or downloading the smart details of the nvme drive.

jyoung8607, 2021-08-16:
Indeed we can see SMART stats from the NVME and it seems okay.

In separate DM discussions with @pd0wm, we think the most plausible explanation is a temporary brownout putting the NVME in a weird state. This can happen due to the C3's backup supercaps and the particular timing of how I unplugged, moved, and repowered the device just before both the events recorded in this issue. Both the brownout and the resulting kernel panic can be mitigated in software, and he's working on that.

RedBaron2, 2021-08-16:
@jyoung8607 
Supercaps can create weird power states for hardware devices. Thanks for keeping this issue updated, and I hope it is rectified via software.

closed 2021-08-21 by adeebshihadeh