commaai/openpilot #21986: Cimma three locked up then rebooted during a drive

brittonx, 2021-08-20:
Running on Comma Three CCE, During a drive, Openpilot stopped steering and throttle. Screen still had green border. Pressing gas and brake did not  remove green border.  About 10 seconds later, the screen went dark then I got the cross country logo. The C3 had rebooted.

adeebshihadeh, 2021-08-20:
Can you fill in the bug report template?

brittonx, 2021-08-20:
Okay.  Where is the template?  I looked here and on discord.

jyoung8607, 2021-08-20:
```
**Describe the bug**

<!-- A clear and concise description of what the bug is. -->

**How to reproduce or log data**

<!-- Steps to reproduce the behavior. -->

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

<!-- Ensure the full logs are uploaded for the posted route. To do so, ensure the Upload Raw Logs toggle is on and connect the device to wifi until all four colums in my.comma.ai/useradmin are filled. -->

**Additional context**

<!-- Add any other context about the problem here. -->
```

Fill in everything you can, but dongle ID is the most critical thing. Based on your description, it's very likely your device gathered and uploaded some useful diagnostic info as soon as it rebooted, but comma will need to know which C3 is yours to look it up.

brittonx, 2021-08-20:
ID: aeaff1b402063ccb

brittonx, 2021-08-20:
Upload raw logs was off.  I just turned it on.

adeebshihadeh, 2021-08-20:
Same issue as #21928. A fix will be pushed to master soon and go out with 0.8.9.
```
[40179.503827] handle_linkdown_irq: PCIe link is down for RC1
[40180.399053] nvme 0001:01:00.0: of_irq_parse_pci() failed with rc=134
[40181.976353] blk_update_request: I/O error, dev nvme0n1, sector 86190256
[40181.976393] Buffer I/O error on device nvme0n1, logical block 10773782
[40181.976407] Buffer I/O error on device nvme0n1, logical block 10773783
[40181.976415] Buffer I/O error on device nvme0n1, logical block 10773784
[40181.976423] Buffer I/O error on device nvme0n1, logical block 10773785
[40181.976430] Buffer I/O error on device nvme0n1, logical block 10773786
[40181.976438] Buffer I/O error on device nvme0n1, logical block 10773787
[40181.976445] Buffer I/O error on device nvme0n1, logical block 10773788
[40181.976452] Buffer I/O error on device nvme0n1, logical block 10773789
[40181.976459] Buffer I/O error on device nvme0n1, logical block 10773790
[40181.976466] Buffer I/O error on device nvme0n1, logical block 10773791
[40181.977561] Unable to handle kernel NULL pointer dereference at virtual address 00000008
```

closed 2021-08-20 by adeebshihadeh