commaai/openpilot #21990: "source code string cannot contain null bytes" on boot

davetapley, 2021-08-20:
**Describe the bug**

Got `source code string cannot contain null bytes` on Comma 2 boot:

[image]

**How to reproduce or log data**

Be unlucky?

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: `32e1dfc5f8dcf575`
 - Route: _I can't remember_ 😞 
 - Timestamp: _I can't remember_ 😞 
 - Version: 0.8.7
 - Car make/model: Toyota Highlander Hybrid 2019

**Additional context**

I [brought this up on Discord](https://discord.com/channels/469524606043160576/524592892627517450/873673331856011334) when it happened but didn't create an issue because I figured it was a one off (and fixed with [full reset](https://github.com/commaai/openpilot/wiki/FAQ?#how-can-i-reset-the-device)), but someone else just [posted the exact same issue in an OP Facebook group](https://www.facebook.com/groups/503673133583617/?multi_permalinks=801952157089045&hoisted_section_header_type=recently_seen), so perhaps not just _cosmic rays_ after all?*

\* and even if is cosmic rays maybe catch `ValueError` and suggest [full reset](https://github.com/commaai/openpilot/wiki/FAQ?#how-can-i-reset-the-device)?

pd0wm, 2021-08-20:
This is rare but known updater bug. For now the best solution is just to factory rest when it happens.

fumark22, 2021-08-22:
After i did a factory reset and reinstalled openpilot.comma.ai. comma 2 is not engaging when I try to turn on openpilot. Also it's stuck on this screen. I can go in the settings but that's it. After 5 min the screen goes black.[image]

pd0wm, 2021-08-22:
It's not connected to the panda for some reason, but that's a different issue. Maybe people on discord can help you debug. If it's a hardware issue contact support@comma.ai.

adeebshihadeh, 2022-04-05:
We suspect this is a bug in the overlay FS implementation in the NEOS kernel. Since we haven't seen this on a comma three, I'm going to close this as the comma two isn't expected to get many more updates.

closed 2022-04-05 by adeebshihadeh