commaai/openpilot #21998: Signal noise from gas pedal cause car disengage randomly.

fraserxiong, 2021-08-20:
**Describe the bug**

OP disengages randomly, or OP disengages immediately after enable.
The issue seems not car specific, but we have seen many Hondas with the same issue. So far, it has been confirmed in Odyssey, Ridgeline, and Pilot. It is also possible that Honda is the most common car brand to use a comma pedal.

**How to reproduce or log data**

Some community member with comma pedal has the bug since 0.8 (I don't know the exact version that causes the issue)
To reproduce the bug, they need to switch back to the official version and engage openpilot. The bug will appear randomly.

**Device/Version information (please complete the following information):**
 - Device: comma two
 - Dongle ID: N/A will get hold with one of the community members to add their data
 - Route:  N/A will get hold with one of the community members to add their data
 - Timestamp: N/A, will get hold with one of the community members to add their data
 - Version: 0.8.4
 - Car make/model: Honda odyssey

**Additional context**

I have made a small change with a higher threshold on disengage openpilot when the gas pedal is not pressed.
Many of the currently affected community members are using my folk. And the change is here: 
https://github.com/fraserxiong/openpilot/commit/ae10bc526a5800fe38c50ed82c89666e344aee65

Also made a wiki about the issue here:
https://github.com/fraserxiong/openpilot/wiki/Noise-with-gas-pedal-issue

jasonmoreau, 2021-08-21:
I am experiencing the same issue in my 2018 Toyota RAV4 XLE (non-hybrid).

antflores, 2021-08-25:
Happens in my 2017 Pilot touring.

gricha01, 2021-11-07:
This happens in my 2019 Honda Passport also. (no forks/giraffes or anything added on)
And this would happen with me on a straight road with no cars around.
Sometimes it takes 3 or 4 times before it stays on. 
I would say 75 percent of the time it works on the 1st try.

adeebshihadeh, 2021-11-21:
@fraserxiong want to open a PR to get this fixed? That threshold is quite a bit higher. We should get a route from each affected car to determine an appropriate new threshold.

fraserxiong, 2022-01-06:
> @fraserxiong want to open a PR to get this fixed? That threshold is quite a bit higher. We should get a route from each affected car to determine an appropriate new threshold.

Thanks for the reply. Will gather data and come up with a reasonable number. Might take a while.

CineXWorm, 2022-01-10:
Not experiencing this on my VW Golf MK6 with comma pedal.

J3ffrey2020, 2022-01-11:
Having the same problem on a ridgelline, this makes the C2 useless if it randomly cancels itself when a car slows down infront of you! Is Comma offering a refund for people effected?

adeebshihadeh, 2022-01-11:
> Is Comma offering a refund for people effected?

Nope, the comma pedal is community hardware, which we do not sell or support.

J3ffrey2020, 2022-01-11:
> > Is Comma offering a refund for people effected?
> 
> Nope, the comma pedal is community hardware, which we do not sell or support.

I should have mentioned mine is doing it with no pedal attached and stock hardware and software.

sshane, 2022-02-23:
Seems like some specific users' cars always have noisy gas interceptor signals, but other vehicles of the same model are fine. https://imgur.com/a/0EwE90y

sshane, 2022-02-23:
Can any of you (@fraserxiong @antflores @gricha01 @J3ffrey2020) provide a route where this happens? Strange that some gas interceptors are this noisy and not following the car-reported gas sometimes.

gricha01, 2022-02-23:
I have V0.8.13 release and it seems as though i am no longer getting this. thanks.

jlundstrom, 2022-02-28:
I've been experimenting with changing the Pedal firmware to sample measurements instead of using point in time measurements for CAN messages and it's been effective at smoothing the values.

For the Panda [SMPR1 (10.13.4)](https://www.st.com/resource/en/reference_manual/rm0033-stm32f205xx-stm32f207xx-stm32f215xx-and-stm32f217xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) is set so that [480 cycles](https://github.com/commaai/panda/blob/master/board/stm32fx/lladc.h#L16) are used for sampling voltage/current (I'm guessing this is done due to noisy results), and the Pedal is using the default of 3 cycles. I'm guessing this is to minimize latency on passing pedal values through which we do not want to compromise on. Comments state there are [65536 cycles](https://github.com/commaai/panda/blob/master/board/pedal/main.c#L310) between each message so we have ample time to add smoothing for the reported values.

I've put together an implementation sampling the last 32 measurements for a ACCEL 1. Driving around and checking the results it seems to have solved this problem for me.  I'm not sure if this is the type of solution we are looking for this problem, but figured I it wouldn't hurt to share.
https://imgur.com/uyxmd9S
https://github.com/jlundstrom/panda/commit/4bda3aaa0179c3359cb7e15b45220bec376d0e67
2205ba1364b32b3e|2022-02-27--15-38-34

sshane, 2022-04-27:
Closing as this seems to be a hardware issue with these specific pedals

closed 2022-04-27 by sshane