commaai/openpilot #22012: openpilot unavailable system malfunction:reboot your device.

Torch-HXM, 2021-08-22:
**Describe the bug**
My carla vision is 0.9.11 and my openpilot vision is 0.8.8.
When I run openpilot in carla,it has a problem of "openpilot unavailable system malfunction:reboot your device",picture as follow:
[image]

**How to reproduce or log data**
My steps are like this:
1.start carla in the first terminal:
[image]

2.start openpilot in the second terminal

3.press '1' and then comes out this issue.

**Additional context**
To solve this problem, I have done some exploring and try. 
I used to stop carla when carla and openpilot were both running,then the problem didn't come out again.And you can press '1' to add the speed as the official documentation says,but the car didn't move at all.In addition,when I stop carla,the ui image came out a logo like this:
[image]

I have no idea about how to solve this problem.I would appreciate it if you can help me.

Operating system: [e.g. Ubuntu 20.04]

riponcse10, 2021-08-27:
All of a sudden, I am also facing the same issue. It worked fine till 15-20 days ago, I don't know whether it is a version compatibility issue.

adeebshihadeh, 2021-08-28:
Should be fixed by 9afa0eec6451a37a0dc0c39bf4837edd2ad53531.

closed 2021-08-28 by adeebshihadeh

Torch-HXM, 2021-08-30:
> All of a sudden, I am also facing the same issue. It worked fine till 15-20 days ago, I don't know whether it is a version compatibility issue.

> Should be fixed by [9afa0ee](https://github.com/commaai/openpilot/commit/9afa0eec6451a37a0dc0c39bf4837edd2ad53531).

thank you for your replay.I solve my problem successfully according to your advice.

Torch-HXM, 2021-08-30:
> All of a sudden, I am also facing the same issue. It worked fine till 15-20 days ago, I don't know whether it is a version compatibility issue.

Thank you.