commaai/openpilot #22030: Another Unprompted Exit

Achilles308, 2021-08-24:
OP took an exit on laneful unprompted. This has been a problem exit in the past, and it seems if there is a lead car, it really struggles and hugs or takes the exit. I believe OP looses track of the lane lines and the lead car blocking the lines doesn't help.

Same issue addressed here #21896

comma 3
master-ci 0.8.8
dfb993cc43ba69dc|2021-08-23--06-57-01--40
07:36::52 local time
2020 Toyota Corolla Hatchback

I had to take over pretty quickly in this instance due to traffic.

Achilles308, 2021-08-25:
Hugged today but did not exit. I could let it do its thing entirely this time, it was sketchy, but didn't exit. I did not touch the wheel any here.

comma 3
master-ci 0.8.8
dfb993cc43ba69dc|2021-08-24--06-56-59--39
07:36::24 local time
2020 Toyota Corolla Hatchback

Achilles308, 2021-09-04:
Same exit, same issue of OP taking the exit. This one I believe was caused by poor lighting. They might all be because of poor lighting come to think of it. The sun is directly ahead and the road is in harsh shadow because of the trees and the other cars and their long shadows. The lines are faded and the seems and cracks in the road are very prominent. 

comma 3
master-ci 0.8.9
dfb993cc43ba69dc|2021-09-03--07-04-43--41 and 42
07:46::39 local time
2020 Toyota Corolla Hatchback

This one was kind of scary as even after I corrected to go straight it veered off again to take the exit still! Very unnerving.

C0mpl3t3N00b, 2021-09-05:
Can confirm also getting this on SA-v0.8.9 _(I understand it is a fork but the model is the same as confirmed by @sshane himself)._

0aa168c5ee8adc43|2021-09-05--16-03-35--11 (16:14:12, 16:15:28, 16:19:46, 16:24:04)

Can be seen both on LaneLess and LaneFull. It has been diving into exits consistently with or without a lead car taking the exit.

Achilles308, 2021-09-10:
My problem exit still took it this morning (and still fought me to try to continue taking it somewhat after I corrected it). I'm tired of uploading it... but thought I'd offer some more input.

closed 2021-09-13 by adeebshihadeh