commaai/openpilot #22031: Since v0.8.5, OP no longer keeps breaking when cruise disengages on Honda

nahcyrag, 2021-08-24:
**Describe the bug**
Prior to v0.8.5, if OP was breaking when cruise disengages, it would continue to brake behind the lead car all the way to 0mph. Since v.0.8.5, it doesn't do that anymore. It just disengages cruise and lets go of the brake. 

my Honda odyssey 2019 would disengages at 15mph and pops up a message saying speed too slow. 
**How to reproduce or log data**

When following a lead car, cruise would disengages at 15mph and stopped braking and sending the car towards to the lead car. 

**Device/Version information (please complete the following information):**
 - Device: Comma 2
 - Dongle ID: e6b5127ea03cc178
 - Route: e6b5127ea03cc178|2021-08-23--19-30-57
 - Timestamp: 19:42:13
 - Version: 0.8.8 stock
 - Car make/model: Honda Odyssey 2019

**Additional context**

this issue was supposed to be fixed in issue #21355. i didn't expect this to happen again but instead of at 22mph now at 15mph.

jedberg, 2021-08-25:
I'm the original reporter of this bug, but I have not upgraded to 0.8.8.  I'm actually still running 0.8.3.  My system is working flawlessly right now, so I'm not super inclined to upgrade to test this.  However, if someone would like to send me a Comma2 with .0.8.8 loaded on it, I'd be happy to go for some test drives and then send the unit back.  Alternatively I'm located in Cupertino, so I'd be happy to head up to SF for the day to help do some testing if you don't have a Honda to test with.

tiki-bug, 2021-08-30:
I experienced the original issue as well and can confirm that for v0.8.6 and v0.8.7 this issue was resolved and that the car steers and brakes down to zero so long as there are no acceleration commands.  Have not upgraded to 0.8.8 yet to confirm the issue's re-emergence.

adeebshihadeh, 2021-09-02:
Thanks for confirming it's working for you @tiki-bug.

@nahcyrag openpilot isn't engaged at all, nor does the car move in the route you posted. Please post a route with the issue and ensure it's fully uploaded.

adeebshihadeh, 2021-09-07:
Closing due to inactivity and others have confirmed this is fixed. If this is still an issue for you, re-open and post a route with the issue.

closed 2021-09-07 by adeebshihadeh