commaai/openpilot #22038: ie. It fingerprints properly.  Still overloaded ``selfdrive/boardd/panda.cc: Transmit buffer full

destinbuons, 2021-08-24:
Dongle id : ecea01e9c860fd76
NEOS 18 
2020 Toyota Avalon trd

ie. It fingerprints properly.

Still overloaded with...
```selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/ui/ui.cc: visionIPC receive timeout
selfdrive/ui/qt/onroad.cc: slow frame time: 109.61
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/ui/ui.cc: visionIPC receive timeout
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/ui/qt/onroad.cc: slow frame time: 108.62
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full
selfdrive/boardd/panda.cc: Transmit buffer full```

ErichMoraga, 2021-08-24:
Reminds me of https://github.com/commaai/openpilot/issues/21470

Apparently @spektor56 recently went through this too...
https://discord.com/channels/469524606043160576/524592892627517450/879582951724159006

pd0wm, 2021-08-24:
When the command to send CAN packets to the panda times out this can either mean the transmit buffer was full (previous messages were not ACKed so it's still retrying), or the transfer timed out for some other reason. In #21470 USB was having other issues that was causing these messages, probably not what's happening in this issue.

@destinbuons Can you be more specific in your bug report? What is working and what's not working? Can you try unplugging *all* connectors and plugging them back in?

destinbuons, 2021-08-24:
I did reconnect all all connectors but it still CAN ERROR CHECK ALL CONNECTIONS

ErichMoraga, 2021-08-24:
@destinbuons Perhaps you can try my recommended right-angle USB 3.1 Gen2 pinned in `#installation-help`

adeebshihadeh, 2021-09-02:
I checked one of your routes, and the canValid flag is false. This is likely a hardware failure. I'd suggest replacing the cable from your device to the harness. If that doesn't work, contact support@comma.ai.

closed 2021-09-02 by adeebshihadeh

ErichMoraga, 2021-09-02:
@destinbuons To be clear, when checking `carState` for a segment in a route, it's common for the first segment (ie. '0') to have `canValid` false for the first 6 seconds or so.  That said, I'm hopeful Adeeb is correct, and it's just a USB cable issue.