commaai/openpilot #22062: Openpilot 0.8.8 triggers immediate LKAS, ACC, and Lane Departure Warning on Civic Hatchback

keijikage, 2021-08-28:
Installing Openpilot 0.8.8 on a C3 hooked up to a 2021 Civic Hatchback causes an immediate LKAS, ACC, and LDW warning.  Re-seated harness with no resolution.

Downgrading to  0.8.7. version resolves issue.

**Device/Version information (please complete the following information):**
 - Device:Comma 3
 - Dongle ID: 546b3e758686e75a
 - Route: 546b3e758686e75a|2021-08-28--16-26-33
 - Timestamp: Immediately on startup
 - Version: 0.8.8 Release 3
 - Car make/model: 2021 Civic Hatchback Sport

May impact more than one honda model, multiple users in discord indicating similar symptoms on C2 hardware as well

keijikage, 2021-08-29:
New route -  546b3e758686e75a|2021-08-29--01-42-44--0
Reset C3 prior to uploading prior logs.

riley206, 2021-08-29:
I'm having the same issue on my 2020 civic hatch. How do you go back to a previous update?

adeebshihadeh, 2021-08-29:
The issue appears to be caused by https://github.com/commaai/panda/pull/615. The new HUD msgs `0x33DA` and `0x33DB` found in the 2021 Accord also appear to be used on the Civic Hatch. #22063 should fix the issue.

riley206, 2021-08-29:
I'm honestly not to sure what all that means yet but I appreciate the response.

closed 2021-08-29 by adeebshihadeh

adeebshihadeh, 2021-08-29:
The fix is in master-ci and will go to the release branches shortly. If you'd like to install master-ci, you can use `openpilot-nightly.comma.ai`.

csouers, 2021-08-30:
@riley206 and @keijikage; When you engage openpilot on 0.8.7 or older and have NOT pressed the LKAS button on the steering wheel, do the LKAS lines show up solid white on the car's display?

keijikage, 2021-08-30:
> @riley206 and @keijikage; When you engage openpilot on 0.8.7 or older and have NOT pressed the LKAS button on the steering wheel, do the LKAS lines show up solid white on the car's display?

The last stock version I ran on any regular basis was 0.8.4, but it did not have this behavior - e.g. the observed behavior is that the lines were off when OP was off (LKAS not engaged).

The only exception to this is when the Comma is booting AND LKAS is engaged prior to fully booting - the car display can become inverted from the expected.

csouers, 2021-08-30:
Would you be able to send me the Copy Share Link from cabana from a working drive? I'd like to take a look.

riley206, 2021-08-30:
> @riley206 and @keijikage; When you engage openpilot on 0.8.7 or older and have NOT pressed the LKAS button on the steering wheel, do the LKAS lines show up solid white on the car's display?

When I engage open pilot on 8.7 a d do not have the lkas button on the white lines on the dash do not appear. When open pilot is engaged and lkas is on, the white lines appear like normal above 45mph. But it asks me to touch the wheel.

PR #22063 (2021-08-29): Revert 2021 Accord
Reverting to fix #22062. Will merge back in after writing tests