commaai/openpilot #22064: Merge Accord 2021 back in

adeebshihadeh, 2021-08-29:
- [ ] write tests for #22062
- [ ] merge accord 2021 back in

adeebshihadeh, 2021-08-29:
@csouers other Bosch models like the Civic Hatch also have `0x33DA` and `0x33DB`, but we just pass them through. Before merging the 2021 Accord back in, I think we should look into whether we should also send those on the Civic Hatch and any other Bosch models that use those.

csouers, 2021-08-30:
Looking into it!

spektor56, 2021-09-01:
2021 CRV Hybrid also has this

austinwylie, 2021-11-01:
Is there a path to getting this back in? Just an Accord 21 owner, looking forward to having full support for the car again. 🙂 Let me know if I can help with drives or anything.

adeebshihadeh, 2021-11-24:
@austinwylie Getting this merged back in with #23021 in the next release that will get out in a few weeks. Once that's merged, you can run master-ci in the meantime, and just confirm everything works as before.

closed 2021-11-24 by adeebshihadeh