commaai/openpilot #22068: Adding additional fingerprints causes openpilot to not find the panda.

dri94, 2021-08-29:
**Describe the bug**

When adding additional 2021 Hyundai Sonata Hybrid fingerprints, openpilot will report "no panda found". If you remove the additional fingerprints, then openpilot reports panda working fine.

**How to reproduce or log data**

Update values.py for the hyundai sonata hybrid with the code below

```
  CAR.SONATA_HYBRID: {
    (Ecu.fwdRadar, 0x7d0, None): [b'\xf1\000DNhe SCC FHCUP      1.00 1.02 99110-L5000         ', ],
    (Ecu.eps, 0x7d4, None): [
      b'\xf1\x8756310-L5500\xf1\000DN8 MDPS C 1.00 1.02 56310-L5500 4DNHC102\xf1\xa01.02',
      b'\xf1\x8756310-L5450\xf1\000DN8 MDPS C 1.00 1.02 56310-L5450 4DNHC102\xf1\xa01.02',
    ],
    (Ecu.fwdCamera, 0x7c4, None): [
      b'\xf1\000DN8HMFC  AT USA LHD 1.00 1.04 99211-L1000 191016',
      b'\xf1\000DN8HMFC  AT USA LHD 1.00 1.05 99211-L1000 201109',
    ],
    (Ecu.transmission, 0x7e1, None): [b'\xf1\000PSBG2323  E09\000\000\000\000\000\000\000TDN2H20SA5\x97R\x88\x9e', ],
    (Ecu.engine, 0x7e0, None): [b'\xf1\x87391162J012\xf1\xa0000R', b'\xf1\x87391162J012\xf1\xa0000S'],
  }
  ```

**Device/Version information (please complete the following information):**
 - Device: comma 2
 - Dongle ID: 0465ff66b61dd65a
 - Route: N/A (bug can be found as soon as you start device)
 - Timestamp: N/A
 - Version: 0.8.8 and 0.8.9 on devel
 - Car make/model: Hyundai Sonota Hybrid 2021

**Additional context**

These fingerprints are needed to support more 2021 sonata hybrids. Not all cars have the same fingerprint as can be seen with other cars such as  the Kia optima, elantra, etc.

adeebshihadeh, 2021-08-29:
Where do you see "no panda found"? There's no user-facing error with that text.

dri94, 2021-08-30:
It says "No panda" to be exact. Here is an image of what it looks like
https://user-images.githubusercontent.com/6392265/120093630-8d91fd00-c0e9-11eb-94fa-f6d0af375988.jpg
@adeebshihadeh

adeebshihadeh, 2021-08-30:
Do you have a link to the branch you're running?

dri94, 2021-08-30:
Yea, its the master branch of my fork. Here is a direct link to the edits https://github.com/dri94/openpilot/blob/5d490f1ce520e75988017e28ece40f261016b51f/selfdrive/car/hyundai/values.py#L849

adeebshihadeh, 2021-09-21:
If this is still an issue, contact support@comma.ai.

closed 2021-09-21 by adeebshihadeh

dri94, 2021-09-22:
@adeebshihadeh I see the EPS was updated, but not the other values. Any reason for this?

joker3287, 2021-09-24:
I just got my comma three and have everything installed and working on my 21 sonata hybrid but i have noticed I get a “no exact car model” warning every time it starts up. Is there a way to fix this? I am really new to this and have been doing a lot of research so I can try to understand what a lot of this stuff means. I appreciate any help