commaai/openpilot #22073: LKAS issue on 2021 Honda CR-V Hybrid

bobbygg, 2021-08-30:
I'm using stock OP 0.8.8. When I hit the LKAS button after OP is engaged using the set button, I get the lkas solid lines on the dash. After a few moments, I get "steering required" on the dashboard, if I don't grab the steering wheel within a few seconds, those solid lines become broken lines. The steering required message stays on until I grab the wheel. My C2 can see my face during this whole ordeal. My understanding is that the LKAS button is not suppose to do anything when OP is engaged.

adeebshihadeh, 2021-08-30:
Please fill out the bug report template so we can look into this.

bobbygg, 2021-08-30:
Where can I find the template?Sent from my Galaxy

spektor56, 2021-08-30:
From the looks of the video he posted: https://www.youtube.com/watch?v=ndoPz-u3iWQ
it looks like the stock wheel nag for having hands off the wheel is being passed through....

pd0wm, 2021-08-31:
@bobbygg 

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

bobbygg, 2021-08-31:
Device: comma two
Dongle ID: 222d53e4e5159f54
Route: 222d53e4e5159f54|2021-08-30--18-51-27
Timestamp: i keep testing it through out the drive. start around 19:01 when i first got on the freeway. 
Version: 0.8.8 release 2
Car make/model: Honda CR-V Hybrid 2021

bobbygg, 2021-09-01:
Route: 222d53e4e5159f54|2021-08-31--16-15-01
This is a cleaner drive. At 16:21, I engage OP, then I hit the LKAS button. The lanes lines show up as solid on the dash. Moments later, "steering required" shows up on the dash. OP was disengaged when I hit the brakes. Then at 16:22, I hit the LKAS button and the lines are broken this time. I then engage OP and the lines remained broken.

pd0wm, 2021-09-01:
Can you make sure the rlogs are uploaded? Turn on your device, go to https://my.comma.ai/useradmin/?onebox=222d53e4e5159f54%7C2021-08-31--16-15-01 and click on "upload all logs", which should show up after a second or two.

bobbygg, 2021-09-01:
I didn't see any "upload all log" option. I hit the "make public" link. Hope that helps.Sent from my Galaxy

spektor56, 2021-09-01:
Looks like it's related to this: https://github.com/commaai/openpilot/pull/22065

His car has messages:

LKAS_HUD_A | 128:33da
LKAS_HUD_B | 2:33db

spektor56, 2021-09-01:
I reverted that change and added his CRV to the extended list and he says its working now.

adeebshihadeh, 2021-09-01:
Let's first get the new HUG msgs from the 2021 Accord PR merged back in for the cars we already support that have them. Then, once that's merged in, we can merge the 2021 Accord back in.

bobbygg, 2021-09-02:
here's a drive i did earlier today on the freeway with the changes spektor56 made.
route # 222d53e4e5159f54|2021-09-01--11-48-57

bobbygg, 2021-09-04:
On route 222d53e4e5159f54|2021-09-03--14-55-02--0, while OP and LKAS is engaged with solid lines, around 16:07, the steering is rocking back and forth until it slowed down to 5mph at 16:10. I then grab the wheel and changed lanes. I let go of the wheel and it resumed shaking until I grabbed the wheel around 16:25. I had the same rocking back and forth before Spektor made the change.  I would get get a solid Lkas lines when LKAS is enaged, then after a few moments I would get the "steering requested" on the dash. If I don't grab the wheel, the steering wheel would shake and the LKAS lines become broken until I grab the wheel or disengage OP.

After Spektor made the change, the lines remain solid like it should. But it seems like some or part of the Honda LKAS system is still interfering with OP. I also notice my vehicle is hugging the left side as if OP is not present. On my 2018 CRV, the vehicle is perfectly centered when OP is engaged.

PR #23000 (2021-12-05): Honda: use new Bosch LKAS_HUD messages
resolves #22073

- [x] <s>determine all cars this applies to</s> auto detect instead
closed 2021-12-05 by adeebshihadeh