commaai/openpilot #22083: Corolla Hatchback - RPM's Hung High

Achilles308, 2021-08-31:
I was driving along with OP engaged and upon a moderate hill OP had the throttle rev really high (like maybe 4,000 RPM?) while the car seemingly was not accelerating to the target speed. If I recall, the set speed was to be either 75 or 72 MPH. It hung at I believe 68 MPH. I jumped on the throttle to disengage OP and regain control of the car and upon reducing the throttle I was able to maintain the speed that hung at like 2,600 RPM's.

This has happened a few other times before, but it is inconsistent as to when it happens (even on the same hill and same set speeds). I've had it hit 5,000 RPM on a tiny little hill trying to hit 55 MPH.

comma 3
master-ci 0.8.9
dfb993cc43ba69dc|2021-08-27--07-14-58--22 and into 23
Sometime after 07:37::40 (it's hard to say for me without looking at the throttle graph)
2020 Toyota Corolla Hatchback

I'm not saying 4,000 RPM's wasn't needed to accelerate in that scenario, but it was odd to be that high and not actually accelerate. If I had to guess, I could accelerate to the set speed at 3,600 RPM's.

HaraldSchafer, 2021-09-22:
I don't really see it not hitting the set speed in the plots
[image]
?

Achilles308, 2021-09-28:
As I recall, this incident had a lead car... so maybe the lead car was slowing down and I couldn't tell. What I remember the most was the RPM was high and it was not accelerating the car or changing gears to be lower. I'm not sure if it was my car or OP but it certainly was different. I don't think it has done it for a few releases so maybe it was fixed with some of the long tweaks.

HaraldSchafer, 2021-09-29:
Yeah maybe they fixed it, just comment here with a route if it happens again!

closed 2021-09-29 by HaraldSchafer