commaai/openpilot #22090: Bad nav UI layout on first-time startup

jyoung8607, 2021-08-31:
**Describe the bug**

The navigation UI is compressed to near-nothing on first-time use. I experienced this today as part of the out-of-the-box experience on a RMA swap. I installed stock openpilot, registered the device, started my Prime trial, and drove normally.

The problem DID NOT go away after cycling ignition, but DID go away after rebooting the device.

**How to reproduce or log data**

I think the condition is attempting to use the nav UI just after registering the device and activating Prime, and it seems to persist until openpilot is restarted. I believe I saw this on registration of my last device as well, but I didn't take notes at the time.

**Device/Version information (please complete the following information):**
 - Device: CCC3
 - Dongle ID: 3cfdec54aa035f3f
 - Route: 3cfdec54aa035f3f|2021-08-31--11-32-41
 - Timestamp: All
 - Version: 0.8.8 release2
 - Car make/model: 2018 Volkswagen Golf R

**Additional context**

Seems to just be a UI layout problem.  Nav is "working"; in that vertical sliver on the far-right side I can see part of the ego-vehicle icon and bits of roads moving as I drive. The diagonal banding on the engaged pic are rolling shutter from my camera or something, they're not perceptible in reality.

[image]

[image]

adeebshihadeh, 2021-09-07:
Unless you can reliably reproduce this, I don't think this has anything to do with the first boot. I saw this on my desk today. Hopefully d5face26a82b717ce694aa848641b879076dfe1f fixes it.

jyoung8607, 2021-09-07:
I don't know of any other way to reproduce. I even had a C3 with a touchscreen problem, false/phantom touches when hot, that caused that C3 to do many hundreds of extra rotations between the onroad UI views and I never saw the problem that way, so I'm not holding my breath for seeing it appear again.

That said, all I have is the time I know I saw it, and the time in the past I think I remember seeing it. Correlation isn't causation, and my feelings aren't hurt if you think you've found the bug elsewhere. I can't deregister my device to try replicating.

pd0wm, 2021-09-07:
We hit the bug on a device that's already set up, so might not be related to that. We think https://github.com/commaai/openpilot/commit/d5face26a82b717ce694aa848641b879076dfe1f might fix it, but haven't been able to reproduce, so can't confirm.

jyoung8607, 2021-09-07:
If I ever see it again, I'll try to get a core with SIGABRT. Till then, doesn't seem like there's any more actions to take.

adeebshihadeh, 2021-09-08:
Looks like this isn't fixed. Hubblesphere reported this on discord while running devel-staging.

adeebshihadeh, 2021-11-08:
@pd0wm isn't this expected to be fixed by a recent change?

jyoung8607, 2021-11-08:
I had an issue earlier today which may or may not be related:

While onroad, engaged and actively navigating, my C3 display briefly reverted to the comma logo and then came back. Driving performance wasn't affected. I haven't looked into it at all, but it felt like a display server or UI watchdog restart.

When the UI came back, the nav layout was entirely missing. In settings it still showed me navigating, and let me clear and set my destination just fine, but the usual three-state onroad UI was missing navigation, reverting to two-state as if I didn't have nav at all. I couldn't get it back even after (re)setting my destination on-device in settings.

I didn't write anything up because I had seen Willem's work in progress to separate the UI and nav processes, and judged it unlikely this class of bug would survive the transition, but thought I'd mention here for S&Gs. I thought it interesting there was another nav layout issue of sorts, and that it was potentially startup order related. Also, tests that check things keep working after late UI starts and multiple UI restarts, because UI watchdog restarts are a thing.

Affected route: `3cfdec54aa035f3f|2021-11-07--10-12-37` (still uploading at this moment)

pd0wm, 2021-11-08:
@adeebshihadeh Yeah, this specific bug should be fixed.

@jyoung8607 We now rely on one of the widgets on the home screen to update the global prime status. I think you hit a UI freeze/watchdog, and the ui restarted. If it starts while onroad, it probably doesn't update this and the ui thought you didn't have prime.

closed 2021-11-08 by pd0wm