commaai/openpilot #22093: unable to download updates?!

Na112786, 2021-08-31:
I can't update my C2. The notes given by alert that : " unable to download updates, Ensure the software is correctly installed", Evidence 1pic. 

Last update check & check for update ,Evidence 2pic + 3pic.

Device: [comma two]
Dongle ID: [1631748f51929063]
Route: [1631748f51929063|2021-07-28--06-22-45]
Timestamp: [31.8.2021]
Version: [release2 (0.8.7)]
Car make/model: [honda pilot 2019]

Thanks.

Na112786, 2021-08-31:
[image]
[image]
[image]

Na112786, 2021-09-02:
Are there any risks of uninstall the openpilot ?
And is it from the same place that I can re-install?

Thanks

adeebshihadeh, 2021-09-02:
No, there's no risks. Simpy install the same way you did when you initially setup the device. Check out #installation-help on the community discord for help.

Na112786, 2021-09-02:
How about, all my driving miles?
Is will delete?

adeebshihadeh, 2021-09-08:
This should largely be fixed by d4b42042331fe797efa144d7e5646c92925386db.

closed 2021-09-08 by adeebshihadeh