commaai/openpilot #22095: Logging/loggerd requirements for training

HaraldSchafer, 2021-08-31:
- All sensor measurements (camera and other sensors) should be accurately timestamped
- Encoder/Loggerd should keep up with camera streams such that no frames are dropped there, if there are, this should raise a flag
- All segments should have synchronized camera streams on C3
- All segments should have same number of frames in 3 camera streams for C3
- All segments should have 1200 frames
- If the camera drops a frame, a blank frame should be inserted and a flag raised

deanlee, 2021-09-08:
>     * [ ]  If the camera drops a frame, a blank frame should be inserted and a flag raised
It is very difficult to do this perfectly in loggerd. and will greatly increase the complexity and performance overhead.
It may be more appropriate to do this in other places, such as framereader

adeebshihadeh, 2021-11-09:
@nuwandavek can you expand the list above with all the checks we do offline?

gregjhogan, 2021-11-09:
> > ```
> > * [ ]  If the camera drops a frame, a blank frame should be inserted and a flag raised
> > ```
> 
> It is very difficult to do this perfectly in loggerd. and will greatly increase the complexity and performance overhead. I think it may be more appropriate to do this in other places, such as framereader or replay

I really want blank frames to easily keep video playback in connect and cabana always in sync with the log data, but now that I think about it more it seems kind of impractical.  For example, an easy case to detect and try to compensate for is when the encoder fails to write out a frame.  But how do we insert a blank frame in?  I assume we have to shove another frame through the encoder which may fail again.  We also don't put anything into the qlog/rlog when a frame is dropped, but again I suppose that can fail, too.

jpc, 2022-03-31:
Hi, I work at Collabora and we invest a lot of work into improving GStreamer – the real-time multimedia framework. Recently we were researching into how people are doing data sampling when uploading videos from embedded devices.

One thing we are curious about is whether it would be useful to have Open Source GStreamer components that could automate the process of: getting the camera data, accurately time-stamping it, running it through a (ONNX?) DL model, compressing it and uploading/streaming it over the internet based on the output from the model.

Since this seems to have a lot of overlap with the problems you are working on I was wondering whether we could compare notes and maybe help each other out?

closed 2022-10-04 by HaraldSchafer