commaai/openpilot #22099: CAN Error: Check connections

bogdan39bikes, 2021-09-01:
**Describe the bug**

**How to reproduce or log data**

**Device/Version information (please complete the following information):**
 - Device: [comma two]
 - Dongle ID: [a08b55b87c4c310b]
 - Route: [a08b55b87c4c310b\|2021-09-01--15-35-28]
 - Timestamp: [about 5-10 seconds after power up]
 - Version: [v0.8.8]
 - Car make/model: [Toyota Camry Hybrid 2021]

**Additional context**

bogdan39bikes, 2021-09-01:
[Turn on the device with all connections connected]

adeebshihadeh, 2021-09-01:
This is most likely a hardware issue. Usually, the cable from the device to the harness breaks. Have you tried to replace that? Also, make sure to connect your device to wifi so the logs can fully upload.

adeebshihadeh, 2021-09-01:
Some of your logs uploaded, and looks like there's no messages on bus 1. This points to a problem with the cable between your harness and device.

closed 2021-09-01 by adeebshihadeh

bogdan39bikes, 2021-09-02:
Today I replaced the usb c as well and the problem still persists

bogdan39bikes, 2021-09-02:
I will make sure that’s it’s on and connected to wifi tonight for the logs