commaai/openpilot #22105: System malfunction alert due to intermittent high CPU

jyoung8607, 2021-09-02:
**Describe the bug**

Driving normally today I had problems engaging with "System Malfunction: Reboot Your Device" reported when I tried. Digging into logs, I see the underlying event was `highCpuUsage`.

**How to reproduce or log data**

Unknown.

**Device/Version information (please complete the following information):**
 - Device: CCC3
 - Dongle ID: 3cfdec54aa035f3f
 - Route: 3cfdec54aa035f3f|2021-09-01--18-15-19
 - Timestamp: All
 - Version: 0.8.8 release3
 - Car make/model: 2018 Volkswagen Golf R

**Additional context**

I was sometimes able to engage during this drive and others on this boot-cycle when this problem was happening, if I caught it when the steering wheel/engageable indicator was flickering on/off in the UI. I did indeed need to reboot to fully clear the problem. I have not looked at logs to see where CPU was going.

pd0wm, 2021-09-02:
Looks like your modem was stuck in the `REGISTERED` state and your device was offline. This caused mapbox to fail loading the tiles, use a lot of cpu and generate a ton of logging. These log messages in turn caused loggerd to use a full core as well.

Every second we retry every failed network request from mapbox, but maybe that wasn't such a good idea in case the device is actually offline. I'll just remove that code for now, since I'm not 100% convinced it was helping anyway.

Is your modem functioning normally on other drives?

jyoung8607, 2021-09-02:
> Is your modem functioning normally on other drives?
It appears not. I noticed it wasn't connected for the drives where I had a problem, but I didn't make a mental connection or include it in the bug, sorry. On another errand this morning it's still not connected. I'm not getting the "NO PRIME" state in the UI either, just offline. I'll look into it later today.

jyoung8607, 2021-09-02:
This seems highly repeatable on my side, so hopefully it should be easy to reproduce and verify a fix on a unit with no mapbox tiles cached and a disabled or removed SIM and no Wifi. In fact, I think the CPU burn may be continuing even after the device goes offroad, judging by unusually ugly startup thermals on a couple errands today away from my home Wifi.

Per our conversation I will start a separate issue for the LTE connectivity issue.

closed 2021-09-06 by pd0wm