commaai/openpilot #22115: No LTE after fresh install on new C3

jyoung8607, 2021-09-03:
**Describe the bug**

After installing release3 openpilot using the normal methods on a factory-fresh C3, LTE connectivity wasn't working. This was discovered and noted as part of a separate issue #22105.

**How to reproduce or log data**

Unknown.

**Device/Version information (please complete the following information):**
 - Device: CCC3
 - Dongle ID: 3cfdec54aa035f3f
 - Route: N/A / all
 - Timestamp: N/A / all prior to late this afternoon
 - Version: 0.8.8 release3
 - Car make/model: 2018 Volkswagen Golf R

**Additional context**

Upon further digging, it was found the default LTE system connection was either never created, or not completely created. In the output below, the UUID-nmmeta symlink to `/dev/null` is of the same UUID as the system default LTE connection.

I don't really know how I got to this state, or if anyone else might conceivably be affected. If so, this does seem something easy to detect and mitigate in startup scripts. I fixed the issue manually by copying the default LTE connection from `/usr/lib/NetworkManager` and restarting NetworkManager. In retrospect, I think it likely a factory reset would also have fixed it.

**Example NetworkManager info from a good C3 with operating LTE:**
```
comma@tici:/data/openpilot$ nmcli conn show
NAME                              UUID                                  TYPE  DEVICE   
openpilot connection young        fa834e4b-2ad9-4ca8-b6bb-013f50914db7  wifi  wlan0    
lte                               6586fb1f-4a66-46d4-9164-0100b73750ee  gsm   cdc-wdm0 
Hotspot                           cbdf53cc-83dc-4539-9161-a7e6aeb0c4be  wifi  --       
openpilot connection jypixel      8939f9c8-ae58-46ed-b12c-6f138ad84980  wifi  --       
openpilot connection STRESS_TEST  27629e64-81f1-4b93-afe8-9956022a34ef  wifi  --  
```

```
comma@tici:/data/etc/NetworkManager/system-connections$ ll
total 24
drwxr-xr-x 2 root root 4096 Aug 11 19:24  ./
drwxr-xr-x 3 root root 4096 May 27 11:16  ../
-rw------- 1 root root  441 Aug 11 19:24  Hotspot.nmconnection
-rw------- 1 root root  363 Aug  2 23:35 'openpilot connection jypixel.nmconnection'
-rw------- 1 root root  308 Jul 30 20:27 'openpilot connection STRESS_TEST.nmconnection'
-rw------- 1 root root  376 Aug  3 00:34 'openpilot connection young.nmconnection'
```

**Example NetworkManager info from this C3 in a bad state:**
```
comma@tici:/data/openpilot$ nmcli conn show
NAME                        UUID                                  TYPE  DEVICE 
openpilot connection young  67419148-9117-4993-bd7f-a5b65a957ebd  wifi  wlan0
```

```
comma@tici:/data/etc/NetworkManager/system-connections$ ll
total 12
drwxr-xr-x 2 root root 4096 Aug 31 17:50  ./
drwxr-xr-x 3 root root 4096 Jul 21 15:00  ../
lrwxrwxrwx 1 root root    9 Aug 31 17:50  6586fb1f-4a66-46d4-9164-0100b73750ee.nmmeta -> /dev/null
-rw------- 1 root root  376 Aug 31 11:16 'openpilot connection young.nmconnection'
```

pd0wm, 2021-09-03:
Looks like something decided to delete the lte connection. It then puts that `.nmmeta` file in `/data/etc/NetworkManager/system-connections` since it can't delete the original file in `/usr/lib/NetworkManager/system-connections`.

```
comma@tici:/data/etc/NetworkManager/system-connections$ l
total 12K
drwxr-xr-x 2 root root 4.0K Aug 30 13:15  .
drwxr-xr-x 3 root root 4.0K Jul 21 12:00  ..
-rw------- 1 root root  393 Aug 30 13:15 'openpilot connection Chansey.nmconnection'
comma@tici:/data/etc/NetworkManager/system-connections$ nmcli con del lte
Connection 'lte' (6586fb1f-4a66-46d4-9164-0100b73750ee) successfully deleted.
comma@tici:/data/etc/NetworkManager/system-connections$ ls -al
total 12
drwxr-xr-x 2 root root 4096 Sep  2 20:09  .
drwxr-xr-x 3 root root 4096 Jul 21 12:00  ..
lrwxrwxrwx 1 root root    9 Sep  2 20:09  6586fb1f-4a66-46d4-9164-0100b73750ee.nmmeta -> /dev/null
-rw------- 1 root root  393 Aug 30 13:15 'openpilot connection Chansey.nmconnection'
```

Not exactly sure what triggered the deletion though, but we should probably clean up these `.nmmeta` files on boot.

jyoung8607, 2021-09-03:
For posterity, the segment where the connection was deleted: `3cfdec54aa035f3f|2021-08-31--17-49-23--1`

As you discovered, that nmmeta marker is how NetworkManager remembers a connection was deleted, and that action is sticky across reboots, so NetworkManager is acting correctly in not bringing up LTE. And that action can be programmatically reversed by removing that nmmeta file shadowing the system default LTE connection.

Question of the day is, why did NetworkManager decide to delete my LTE connection? The fact it did so within 150ms of me departing my home Wifi coverage seems far too close to be coincidence. Still looking on my side as well.

jyoung8607, 2021-09-03:
Per our discussion, openpilot `ui` asked for the deletion. Haven't looked at potential causes yet, but I'm assuming we got in a race somewhere with networks going up and down at the same time.

`[21926.957837] MAIN 3439 NetworkManager - <info>  [1630446633.7613] audit: op="connection-delete" uuid="6586fb1f-4a66-46d4-9164-0100b73750ee" name="lte" pid=48811 uid=1000 result="success"`

`      (name = "ui", pid = 48811, running = true, exitCode = 0)`

jyoung8607, 2021-09-03:
Recapping the last of our discussion from last night. This is my best guess as to where the problem might lie. I don't have cycles to pursue this one further; @pd0wm has it from here.

When I lose Wifi connectivity due to range, it tries to reconnect and fails the auth handshake, creating a state change event to need-auth with reason supplicant-disconnect:

`[21926.801351] MAIN 3439 NetworkManager - <info>  [1630446633.7061] device (wlan0): state change: activated -> need-auth (reason 'supplicant-disconnect', sys-iface-state: 'managed')`

That may be triggering this section of code that assumes it's a bad Wifi password and forgets the connection:
https://github.com/commaai/openpilot/blob/308a584defcb4128efcace0212cea48cfae5a108/selfdrive/ui/qt/offroad/wifiManager.cc#L304-L308

Above is one of the very few callers to forgetConnection(), which is the only place I see a DBus call to delete a connection:
https://github.com/commaai/openpilot/blob/308a584defcb4128efcace0212cea48cfae5a108/selfdrive/ui/qt/offroad/wifiManager.cc#L245-L251

I can't immediately see how that would end up accidentally forgetting the LTE connection instead of the Wifi connection, but it seems an extraordinary coincidence.

adeebshihadeh, 2021-09-03:
- [ ] clear nmmeta files on startup
- [ ] fix wifiManager.cc

twilsonco, 2021-09-03:
I experienced this today and removed the `.nmmeta` file as advised by jyoung8607. Dongle id c11fcb510a549332

jyoung8607, 2021-09-04:
Confirmed same event signature on @twilsonco's device: `c11fcb510a549332|2021-09-01--07-16-59--0`:

```
[77687.226043] MAIN 3865 NetworkManager - <info>  [1630502260.9542] device (wlan0): state change: activated -> need-auth (reason 'supplicant-disconnect', sys-iface-state: 'managed')
[77687.226644] MAIN 3865 NetworkManager - <info>  [1630502260.9545] dhcp4 (wlan0): canceled DHCP transaction
[77687.226772] MAIN 3865 NetworkManager - <info>  [1630502260.9545] dhcp4 (wlan0): state changed extended -> done
[77687.226883] MAIN 3841 dbus-daemon - [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.4' (uid=0 pid=3865 comm="/usr/sbin/NetworkManager --no-daemon " label="kernel")
[77687.227229] MAIN 3865 NetworkManager - <info>  [1630502260.9627] device (p2p-dev-wlan0): supplicant management interface state: 4-way handshake -> disconnected
[77687.227403] MAIN 3865 NetworkManager - <warn>  [1630502260.9642] device (wlan0): no secrets: No agents were available for this request.
[77687.227557] MAIN 3865 NetworkManager - <info>  [1630502260.9642] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
[77687.227678] MAIN 3865 NetworkManager - <warn>  [1630502260.9676] device (wlan0): Activation: failed for connection 'openpilot connection Kingdom'
[77687.227829] MAIN 1 systemd - Starting Network Manager Script Dispatcher Service...
[77687.227942] MAIN 3865 NetworkManager - <info>  [1630502260.9743] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
[77687.228486] MAIN 0 kernel - R0: [soft_irq][1491749937124] [13:17:40.975471]  wlan: [0:E :SME] sme_8023_multicast_list: Invalid session: 0
[77687.229016] MAIN 0 kernel - IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[77687.229148] MAIN 3865 NetworkManager - <info>  [1630502260.9984] policy: set 'lte' (wwan0) as default for IPv4 routing and DNS
[77687.229304] MAIN 3865 NetworkManager - <info>  [1630502261.0024] device (cdc-wdm0): state change: activated -> deactivating (reason 'connection-removed', sys-iface-state: 'managed')
[77687.229431] MAIN 3865 NetworkManager - <info>  [1630502261.0029] manager: NetworkManager state is now DISCONNECTING
[77687.229543] MAIN 3865 NetworkManager - <info>  [1630502261.0062] audit: op="connection-delete" uuid="6586fb1f-4a66-46d4-9164-0100b73750ee" name="lte" pid=46254 uid=1000 result="success"
```

Hubblesphere, 2021-09-04:
I have the same issue as well. Followed jyoung's steps and got LTE connection back.

PR #22129 (2021-09-07): Fixes for C3 LTE connection being deleted
**Description**

Partially resolve #22115 by fixing two `wifiManager.cc` code paths that could delete the LTE connection. 

The first fix will make it so we don't wrongly try to forget connections outside of `ui`'s new connection process. It was possible to trigger this if wpa_supplicant tried to re-authenticate to an existing Wifi network while driving out of range. In that case, it didn't even try to delete the current Wifi connection; instead it was trying to delete an empty SSID connection.

The second fix will make sure the LTE connection can't po [... cut]
closed 2021-09-07 by pd0wm