commaai/openpilot #22132: C3 doesn't honor MAX_TIME_OFFROAD_S

ntegan1, 2021-09-05:
**Describe the bug**

Myself and other users have noticed that the C3 often turns off before `MAX_TIME_OFFROAD_S` is reached.
It has been discussed today and yesterday in #hw-three.
I've been leaving my device plugged in overnight on my desk and in the morning it is almost always "off". There is a blinking blue led on the back but the screen is unresponsive and the device is not seen on my network.

**How to reproduce or log data**

Set `MAX_TIME_OFFROAD_S` to something like 48 hours and leave it for around a day.

**Device/Version information (please complete the following information):**
 - Device: comma three
Don't think logs would help with this one?

**Additional context**

jyoung8607, 2021-09-06:
> I've been leaving my device plugged in overnight on my desk and in the morning it is almost always "off". There is a blinking blue led on the back but the screen is unresponsive and the device is not seen on my network.

How specifically are you powering it? C3/Panda have logic to look for a car harness/relay box on the other end of the OBD-C and should be entirely disabling auto power-down if it sees it's not installed in a car.

ntegan1, 2021-09-06:
> How specifically are you powering it? C3/Panda have logic to look for a car harness/relay box on the other end of the OBD-C and should be entirely disabling auto power-down if it sees it's not [...]

Powering via Panda Jungle so 12V OBD-C. Others have experienced the same issue with regular usb power blocks.

pd0wm, 2021-09-06:
Both C2 and C3 count up power used, which persists across reboots. If the threshold of used power is reached it will shut off, but stay on for a minimum of 1 hour (https://github.com/commaai/openpilot/blob/master/selfdrive/thermald/power_monitoring.py#L22). The only way to get the power used back down is to be on-road. So if you keep powering it on a desk, and the shutoff threshold is reached I would expect it to always shut down in an hour.

closed 2021-09-06 by ntegan1