commaai/openpilot #22134: Unlogger with remote data

fsalsubaei, 2021-09-05:
After a successful authuntication, Unlogger with remote data gives me this error
```
keyboard waiting for input
Process Process-1:
Traceback (most recent call last):
  File "/home/ff/.pyenv/versions/3.8.5/lib/python3.8/multiprocessing/process.py", line 315, in _bootstrap
    self.run()
  File "/home/ff/.pyenv/versions/3.8.5/lib/python3.8/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "unlogger.py", line 69, in run
    route = self._process_commands(cmd, route, pub_types)
  File "unlogger.py", line 146, in _process_commands
    self._lr = MultiLogIterator(route.log_paths(), wraparound=True)
  File "/home/ff/openpilot/tools/lib/logreader.py", line 20, in __init__
    self._first_log_idx = next(i for i in range(len(log_paths)) if log_paths[i] is not None)
```

and this is what I get when running this code
```
from tools.lib.logreader import LogReader
from tools.lib.route import Route
route = Route('3533c53bb29502d1|2019-12-10--01-13-27')
log_paths = route.log_paths()
print(log_paths)
```
result:
`[None, None,, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None, None]`

pd0wm, 2021-09-06:
Rlogs are not uploaded (https://my.comma.ai/useradmin/?onebox=3533c53bb29502d1%7C2019-12-10--01-13-27). Turn on your device, and click the "upload all rlogs" button.

closed 2021-09-06 by pd0wm

fsalsubaei, 2021-09-11:
Thank you.
I did not find the button.
Also, it would be nice to add a message to indicate that instead of the error