commaai/openpilot #22146: Honda Odyssey 2020, long API not fully understood

HaraldSchafer, 2021-09-06:
We use PCM_GAS value to set max acceleration for Honda NIDECs, this does not work for 2020 Odyssey. We need to understand the API for this NIDEC and cars like it.

BBBmau, 2021-09-09:
Hi all, would love to work on this issue. Just needing help in locating the API that you are mentioning. Thanks in Advance!

belm0, 2021-09-24:
> We use PCM_GAS value to set max acceleration for Honda NIDECs, this does not work for 2020 Odyssey. 

Do you mean PCM_ACCEL?  In what way is the Odyssey PCM_ACCEL different than other Hondas?

(I assume that merged PR #22156 was the workaround for this issue.)

belm0, 2021-09-24:
I see, it's PCM_GAS

https://github.com/commaai/openpilot/blob/3ccf712ac99ce8ef393d87ddd7241d1d823dd5a6/selfdrive/car/honda/hondacan.py#L122

this comment threw me off, referring to "PCM_ACCEL":

https://github.com/commaai/openpilot/blob/3ccf712ac99ce8ef393d87ddd7241d1d823dd5a6/selfdrive/car/honda/carcontroller.py#L189-L190