commaai/openpilot #22159: Recently Updated controlsd.pid causes Openpilot from engaging

Red7Hunter, 2021-09-07:
I recently updated to 0.8.9 from 0.8.8  and was unable to get openpilot to start after that.

I was able to confirm CAN BUS messages in both direction.

While engaging I noticed that highCpuUsage was blocking it from starting.

The error on screen is:
System Malfunction: Reboot Your Device

I commented out these lines in controlsd.pid:

    cpus = list(self.sm['deviceState'].cpuUsagePercent)[:(-1 if EON else None)]
    if max(cpus, default=0) > 95 and not SIMULATION:
      self.events.add(EventName.highCpuUsage)

Ubuntu 20.04

i5 NUC with 4 hyperthreading cores.

adeebshihadeh, 2021-09-07:
It sounds like you need more CPU overhead, otherwise you can experience degraded performance. If this is acceptable for you, then feel free to remove the check locally or in your own fork.

closed 2021-09-07 by adeebshihadeh

Red7Hunter, 2021-09-07:
@adeebshihadeh ,

Thanks for the quick reply, but unfortunately that isn't the case.
1) Software was working as designed before change was implemented
2) The threshold for this error is set to 95%.. but 95% of what?  I have multiple cores and their average isnt above 95% some of the are higher or lower.  How is this being measured and is it being accurately measured/defined?
3) The splash screen isn't helpful at all, there appears to be a bunch of errors checks in the controlsd.pid that doesn't provide any user to help diagnose.. examples:  CPU, TEMPERATURE, FAN SPEED, ETC.  generic spash screen isn't helpful for resolution.

Please see my screen shot of CPU usage:
[image]

Let me know if you would like any more data points to improve the master vs a fork.

adeebshihadeh, 2021-09-07:
That check is for per-core CPU usage, not the overall CPU usage. In your screenshot, core 8 is at 100%. If a single core is saturated, this indicates an openpilot process is potentially starved of CPU time.

Red7Hunter, 2021-09-07:
I'll dig in more and report back what process that is.

But that is making an assumption that the process is pegged is related to work being done by autopilot and thus starving it. 

There is potential that no openpilot resources are on that processor and declare a fault state because 1 pressor is above a threshold vs an average seems to void the point of multithreading.

adeebshihadeh, 2021-09-07:
Yes, we assume the device is dedicated to running openpilot, such as a comma two or three. I don't really want to treat PCs any differently.

Red7Hunter, 2021-09-07:
I understand.  I'll confirm what that process is and report back.  

Do you think there could ever be a situation where it's at 100% on purpose?  As in for GPS timing or CAM bus talking... its not actually starved but more dedicated to the task?

dorkmo, 2021-09-10:
wonder if you could overclock your processor? would be interesting to see what happens.

Red7Hunter, 2021-09-11:
I did dig into this more I'll try to upload more info.  

But it doesn't look like it's a process in ubuntu .. I wonder if it has something to do with integrated graphics card being ran on the it?