commaai/openpilot #22189: Comma Three will not turn on in car or with just USB power from computer of aux Charging

littletommysell, 2021-09-10:
**Describe the bug**

**How to reproduce or log data**

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

**Additional context**

grekiki, 2021-09-10:
USB-charging from a computer is very unlikely to provide enough power to boot a C3. Could you provide more details about what you are trying to do exactly?

adeebshihadeh, 2021-09-10:
This is a hardware issue. Contact support@comma.ai.

closed 2021-09-10 by adeebshihadeh

littletommysell, 2021-09-10:
Thanks for reaching out.  My Comma three installed fine in my 2021 Camry and worked for two days.  I now have a black screen with no activity on it.  I then took the Comma into my house and tried the power out let on my Computer and then two different power sources that I used in programing the Comma in my house before installing it in the car.  No Screen activity on any of the Four different power sources.