commaai/openpilot #22198: Comma Three Release 0.8.8 0 2021 Hyundai Palisade car alert error.

llck80ll, 2021-09-11:
Hi, just earlier this morning, the red line (laneless) was being off centered to the left. I uninstalled and reinstalled OpenPilot.

Now, every time I am in the car, I get an alert on my dash saying "Check Forward Collission-Avoidance Assist system" and it keeps beeping.

Is this a known bug/issue being worked on? 

Can send pics, videos, dongleID if necessary.

Thank you for this great technology that has changed my life and how I drive.

pd0wm, 2021-09-11:
Please fill out the bug report template:

**Device/Version information (please complete the following information):**
 - Device: [e.g. EON/EON Gold/comma two]
 - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device -> Dongle ID or my.comma.ai/useradmin]
 - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in my.comma.ai/useradmin]
 - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the drive)]
 - Version: [commit hash when on a non-release branch, or version number when on devel or release2 (e.g. 0.7.6)]
 - Car make/model: [e.g. Toyota Prius 2016]

llck80ll, 2021-09-12:
- Device: [e.g. EON/EON Gold/comma two]

Comma Three

   - Dongle ID: [e.g. 77611a1fac303767, can be found in Settings -> Device
   -> Dongle ID or my.comma.ai/useradmin]

Dongle ID:  29c3af39bd48a377

   - Route: [e.g. 77611a1fac303767|2020-05-11--16-37-07, can be found in
   my.comma.ai/useradmin]

29c3af39bd48a377|2021-09-11--19-25-46

   - Timestamp: [When in the route the bug occurs (e.g. 4min 30s into the
   drive)]

The issue occurs as soon as I start the car and Comma Three is in use or
not. Goes away when I unplug the device.

   - Version: [commit hash when on a non-release branch, or version number
   when on devel or release2 (e.g. 0.7.6)]

Version release 0.8.8

   - Car make/model: [e.g. Toyota Prius 2016]

2021 Hyundai Palisade. Sometimes I also get a message saying the closest
model is the 2020 Hyundai Palisade. It did not used to do that.

llck80ll, 2021-09-12:
I’m going to upload some before and after videos. I think I fixed it by unplugging everything and just re-plugging and working the harness around on the cover. I went on two test drives and the error hasn’t come back yet.

I’m going out with the family on the long drive so hopefully it’s truly fixed by unplugging and replug in the harness in the OBDC see connector. I’ll report back to the ticket at the end of the day.

llck80ll, 2021-09-12:
Here is a video with the car alert in action.
https://youtu.be/UpjWahxHXgY

Here is a video with the alert and then unplugging the device
https://youtu.be/rk1T4eme0DI

Two videos of test drives after I unplugged the OBDC and harness and reconnecting the harness multiple times to make sure there was a good connection.

https://youtu.be/BJiWzjzICe0
https://youtu.be/NIpXQKTxXRc

FYI, the videos are still processing so should be available within the hour.

llck80ll, 2021-09-12:
I THINK we can consider this issue closed. I was out all day driving local/highway and the car alert did not come back.  Here is a video of my most recent drive to the market and back.

https://youtu.be/VxiR4iyBruo

As I said earlier, I disconnected the OBDC and harness multiple times and that may have done the trick?

Thank you, Comma Team, for this revolutionary tech. It does look like you will win the self driving car race and I look forward to the future.

closed 2021-09-12 by adeebshihadeh