commaai/openpilot #22254: Locationd not using carState packets as expected, possible timing issue

HaraldSchafer, 2021-09-16:
In this segment
` 09c32a592a49b2da|2021-08-04--13-41-56/18 `

The carState.standstill flag is always true. This should cause serious issues with the locationd that expects this value to be correct and enforces standstill when it is true. Somehow locationd did not react to this at runtime at all (it does in offline reruns). I suspect there are timing issues between the messages causing messages to be dropped, but unconfirmed.

nuwandavek, 2021-10-05:
This is not a bug. The `valid` attribute of this segment's `carState` message is always `False`. So, it fails this check https://github.com/commaai/openpilot/blob/1cfabc3388443a5480310930e12331d508c92bdd/selfdrive/locationd/locationd.cc#L453

So, the `carState` message is never used to update the filter.

HaraldSchafer, 2021-10-05:
Ok this is just a car bug then adding to this
https://github.com/commaai/openpilot/issues/22253

closed 2021-10-05 by HaraldSchafer