commaai/openpilot #22265: Car crashes on autopilot

pkr-20, 2021-09-17:
**Describe the bug**

faulty autopilot - the car crashes within a couple of seconds after engaging autopilot, even though there is no reason for it to do so 

**How to reproduce or log data**

start the CARLA docker container with `run --rm --net=host -e DISPLAY= -e DSL_VIDEODRIVER=offscreen -it --gpus all carlasim/carla:0.9.11 ./CarlaUE4.sh -opengl -nosound -quality-level=Low -world-port=2000`

run ~/openpilot/tools/sim/start_openpilot_docker.sh

hit '1' to engage autopilot, and watch as the car eventually drives into the side guardrail, unable to move after that. (throttle will max out at 1, and steer will move slightly)

Operating system: Ubuntu 18.04.5

Ming2888, 2022-03-01:
Is there any update on this issue? I encountered the same problem.

maykonpacheco, 2022-03-01:
Why do you need to use
"run --rm --net=host -e DISPLAY= -e DSL_VIDEODRIVER=offscreen -it --gpus all carlasim/carla:0.9.11 ./CarlaUE4.sh -opengl -nosound -quality-level=Low -world- port=2000" instead of "./start_carla.sh" and "./start_openpilot_docker.sh" ?

https://github.com/commaai/openpilot/blob/master/tools/sim/README.md

Ming2888, 2022-03-01:
I am using "./start_carla.sh" and "./start_openpilot_docker.sh", and still have the same issue.

maykonpacheco, 2022-03-01:
Add here an image of the error that is appearing, that way we can see what you are doing.

Ming2888, 2022-03-02:
[image]
[image]

Ming2888, 2022-03-02:
Autopilot does not work, and the vehicle drove off road.

closed 2022-04-20 by pd0wm