commaai/openpilot #22267: Oscillation in velocity and car doesn't reach the set speed

sshane, 2021-09-18:
**Describe the bug**
I don't know when this began, but on a drive tonight I noticed significant oscillation in the speed when I engage at any speed below 40 mph or so. The desired accel and speed from the planner look okay, so this is something in longcontrol.py most likely. This segment was above the minimum ACC speed for pedal users, so that shouldn't be the issue.

**How to reproduce or log data**
Engage at 30 mph-ish on a car without a pedal and you should see oscillation traveling on a flat road.

**Device/Version information (please complete the following information):**
 - Device: comma three
 - Route: f15e3c37c118e841|2021-09-18--01-10-10
 - Timestamp: segment index 0-1
 - Version: master `dc076baf39c8dbf8a7c0ea0e9b06c65b73f6e7ae`
 - Car make/model: 2017 Toyota Corolla

[image]

cydia2020, 2021-09-18:
Same here, observed on my Prius 2018

xps-genesis, 2021-09-20:
same thing observed on Chrysler pacifica. always stays below set speed (see it below 2 mph). did not see this on release 2

HaraldSchafer, 2021-09-20:
@cydia2020 @xps-genesis Please post routes and timestamps, I can't debug words.

HaraldSchafer, 2021-09-20:
@sshane I'm quite confused looking at the data it looks like the PID is just overcontrolling, but not sure how this got introduced. Nothing looks suspicious in the commits. Will keep looking.

sshane, 2021-09-21:
@HaraldSchafer I believe this commit introduced this issue: https://github.com/commaai/openpilot/commit/66c275b71128355d634d4daf0f5c18891d4fb047

I can upload routes of before and after the commit if needed

HaraldSchafer, 2021-09-21:
@sshane Yeah if you have a route from before in the same situation, flat road achieving that set speed with no lead, that'd be helpful!

sshane, 2021-09-21:
Where I just tested it, it's not the exact situation (a bit more hilly) so I'll get it tomorrow and add it here!

cydia2020, 2021-09-22:
I got two drives that are currently being uploaded, will add them here soon

cydia2020, 2021-09-22:
Before acados change: 399b1cca765e80b4|2021-09-22--04-17-10
After acados change: 399b1cca765e80b4|2021-09-22--04-35-34

HaraldSchafer, 2021-09-22:
@cydia2020 Ah you're talking about not reaching the set speed? That was a bug.

Fixed here :)
https://github.com/commaai/openpilot/commit/fcda55fbf1371e95c82755da2069ccb547d47983

cydia2020, 2021-09-23:
> @cydia2020 Ah you're talking about not reaching the set speed? That was a bug.
> 
> Fixed here :)
> [fcda55f](https://github.com/commaai/openpilot/commit/fcda55fbf1371e95c82755da2069ccb547d47983)

Yes, I was mainly talking about that since the title of the issue is indeed "Oscillation in velocity and car doesn't reach the set speed" :)

However I've also experienced the oscillation issue on more hilly roads, especially when going down hill openpilot doesn't seem to know how to keep the car close to it's set speed and keep applying the brakes.

If you need a route of this happening I'm happy to provide that.

xps-genesis, 2021-09-23:
> Yes, I was mainly talking about that since the title of the issue is indeed "Oscillation in velocity and car doesn't reach the set speed" :)
>
> However I've also experienced the oscillation issue on [...]

@HaraldSchafer @cydia2020 the hill stuff will never be smooth because op long controller needs to know the road slope in order to shift the zero point on the accel. 
 
if the car is going uphill, the feed fwd accel term should be enough to manage gsin(θ) that is slowing down the vehicle, to maintian constant set speed.
same thing while going downhill  the accel 0 point needs to shift  to manage the downward pull that is speeding up the vehicle.

the controller may just be fine without the hill influence factor, because these oscillations need long downhill and uphill surfaces to really feel annoyed and it will manage that everything is closed loop, but it will never be smooth and precise.

I hope comma works on adding hill influence factor to its long controller. because a well tuned OEM ACC most probably uses the hill influence in their controller.

sshane, 2021-09-27:
@HaraldSchafer Sorry I've been busy, do you still want a route from before the commit that started this bug?

HaraldSchafer, 2021-09-29:
@HaraldSchafer Just lemme know if you still have issues. The not reaching set speed should be fixed.

And PID values for your car are very high. I recommend trying the milder tune
https://github.com/commaai/openpilot/blob/85dc43d166f6ad8286dfbed39c609959195fbac0/selfdrive/car/toyota/interface.py#L323

cydia2020, 2021-09-29:
I can confirm the not reaching the set speed issue has been fixed

HaraldSchafer, 2021-09-29:
@cydia2020 thanks!

sshane, 2021-09-30:
Haven't seen it have the issue where it doesn't reach the set speed! It still overcorrects at low speed though, so I'll try that improved tune. I tried it before the new MPC (+3 months ago) and my car had trouble keeping the set speed on highways, but the new MPC seems to be a bit more aggressive at planning a trajectory back to the desired speed? Maybe it will be okay now, I'll report back