commaai/openpilot #22292: No Panda Detected

vietsatillite, 2021-09-21:
Hello, 

I have recently run into issues with my comma 2 on my 2019 Toyota Rav4 Hybrid. Openpilot was working fine a few weeks ago prior to an update v0.8.7. It started off with a communication error eventually leading to "CAN Error: Check Connections" and finally No Panda detected was showing up under v0.8.8. The comma 2 has power from the OBD and factory cruise control works fine with the harness installed. I updated the device to v0.8.9 hoping it was a software issue, no change. I also tried using a different Fork, SA-master, to see if a different fork might resolve the issue, no change. 

I contacted comma support and they advised to troubleshoot with a known working panda and harness. I purchased a new one and ran the whole setup up again with the new harness, however, no difference. I still get no panda detected on my comma 2 and have tried driving to check if it would detect a signal down the line, but no dice. 

 - Device: Comma2
 - Dongle ID: 75955AA5F3CA2F0A
 - Route: N/A
 - Timestamp: N/A
 - Version: v0.8.8 & v0.8.9
 - Car make/model: 2019 Toyota Rav4 Hybrid XSE

robbederks, 2021-09-21:
It looks like there's something going wrong in libusb. There have been multiple crashes of boardd, and also there are a lot of `libusb initialization error` logs for this dongle_id.

Unless @adeebshihadeh thinks otherwise, I think this looks more like some kind of HW failure to me, in which case you should probably contact support@comma.ai.

adeebshihadeh, 2021-09-24:
It's strange because the logs indicate it connects somewhat regularly and successfully flashes. This is most likely a hardware issue though.

closed 2021-09-24 by adeebshihadeh