commaai/openpilot #22298: Comma 3 Low Memory

mogorman, 2021-09-21:
**Describe the bug**
Was driving yesterday and got a low memory restart needed warning.  I checked memory usage in htop, majority being used by _ui and 3.1 gigs of memory used total.

**How to reproduce or log data**
Unclear, unit had been on for a little over a day.

**Device/Version information (please complete the following information):**
 - Device: Comma 3
 - Dongle ID: d544af690774f394
 - Route: d544af690774f394|2021-09-21--13-52-23
 - Timestamp: Unclear when error message happened spouse was driving car
 - Version:  b40d51ce2a3052c3bf549d457cd92d9a4ff19989 mog_beta of my fork.  it is based on 0.8.9 devel
 - Car make/model: Chrysler Pacifica Hybrid 2021

**Additional context**

adhintz, 2021-09-22:
My device currently has this warning. Please let me know if you want any data from it.

adeebshihadeh, 2021-09-22:
@adhintz uploading, preserving, and posting the route should be enough

adhintz, 2021-09-22:
preserved it and the route is 1c28f817e18cf5bc|2021-09-22--15-59-46--0

mogorman, 2021-10-01:
had it happen again to me only thing interesting in logs was 
```
Traceback (most recent call last):                                                                                                                                                 
  File "/data/openpilot/selfdrive/thermald/thermald.py", line 273, in thermald_thread    
    if TICI and (network_info.get('state', None) == "REGISTERED"):                                                                                                                 
AttributeError: 'NoneType' object has no attribute 'get'                                                                                                                           camerad clocksd dmonitoringmodeld logcatd loggerd modeld proclogd sensord ubloxd ui soundd locationd calibrationd controlsd deleter dmonitoringd logmessaged pandad paramsd planner
d radard thermald timezoned tombstoned updated uploader                                                                                                                            
selfdrive/loggerd/loggerd.cc: camera 0 rotate encoder to /data/media/0/realdata/2021-10-01--09-12-32--4                                                                            
selfdrive/loggerd/loggerd.cc: camera 2 rotate encoder to /data/media/0/realdata/2021-10-01--09-12-32--4
selfdrive/loggerd/loggerd.cc: rotated to /data/media/0/realdata/2021-10-01--09-12-32--4                                                                                            
selfdrive/loggerd/loggerd.cc: camera 1 rotate encoder to /data/media/0/realdata/2021-10-01--09-12-32--4             
```

closed 2021-12-06 by adeebshihadeh