commaai/openpilot #22356: Car Rendered Invisible?!

Achilles308, 2021-09-27:
I'm not sure what this issue was... but it was a strange one.

I was driving along and this black Kia Soul with rear wreck damage and no lights entered my lane in front of me around dawn. Openpilot was engaged and it did not detect this car at all it seems. It would accelerate through the car if I hadn't intervened. I ended up approaching this car 3 times and each time it did not detect it. This particular bug is extremely dangerous as it would have rammed this car without manual intervention.

First incident:
comma 3
master-ci 0.8.8
dfb993cc43ba69dc|2021-09-24--06-53-09--37 through 40
07:30::33
2020 Toyota Corolla Hatchback

Second incident:
07:32::12
the car was stopping for this turn/exit and OP was not braking appropriately

Third incident:
07:33:25
I ran up the set speed to catch the car... and OP was accelerating very rapidly like the car wasn't there.

Not sure if this car is special or if something in the model changed... but this behavior was very uncharacteristic and worrisome. Had this car been stopped at a red light and had a driver been completely distracted (shame on them!) OP would have happily crashed into it at full speed.

Note to everyone, you must be ready to take over at all times!

pd0wm, 2021-09-28:
Remember openpilot is a level 2 system, and you can never rely on it to take action. The driver should always pay attention and be ready to take over. If you describe these situations as "extremely dangerous" you should adjust your expectations or stop using openpilot.

@HaraldSchafer Can you look into why the model did not detect the lead car?

HaraldSchafer, 2021-09-29:
The model detected it, it appears the lead MPC got into an unstable state and did not recover.

This should be fixed since
https://github.com/commaai/openpilot/commit/fe983a7b8ce7ad63c73663284d8b8e093a280308

This new MPC doesn't have these unstable modes. And if it did go unstable it should now do a full hard reset.

closed 2021-09-29 by HaraldSchafer

Achilles308, 2021-09-29:
> The model detected it, it appears the lead MPC got into an unstable state and did not recover.
> 
> This should be fixed since [fe983a7](https://github.com/commaai/openpilot/commit/fe983a7b8ce7ad63c73663284d8b8e093a280308)
> 
> This new MPC doesn't have these unstable modes. And if it did go unstable it should now do a full hard reset.

Would you be able to run the video and radar data into the simulation with the new model? Just to be sure??

Achilles308, 2021-09-29:
Also, it would be nice if there was an alert when openpilot knows something is wrong... could really save someone's life you know?